betaflight/betaflight #10343: BF4.3 - VTX Control Broken (Tramp)

akfpv, 2020-11-12:
**Describe the bug**
Recent builds of BF4.3 (tested on a build from Oct 17) have Tramp VTX protocol broken (don't know if SA is working). When plugging in a quad the vtx seems to be starting in Pit mode, then for a moment transmitting on the set power, then going into Pit mode again, and after a few seconds transmitting again. This behaviour is repeatable on all quads using different FC's (a mix of F4's and F7's) with Tramp VTX (don't use any Unify's).
Most likely it's been broken in PR #9638 or PR #10166 - tagging @pgreenland for awareness

**To Reproduce**
Steps to reproduce the behavior:

1. flash 4.3 from Oct 17 or later
2. observe VTX not transmitting straight away and intermittently breaking up.

**Expected behavior**
With no FC Pit Switch used VTX should be transmitting right after you plug in the battery

**Flight controller configuration**
```
Entering CLI

# diff

# version
# Betaflight / STM32F405 (S405) 4.3.0 Oct 17 2020 / 07:54:37 (539d0e1e8) MSP API: 1.44
# config: manufacturer_id: FFPV, board_name: FF_RACEPIT_MINI, version: 6fac4231, date: 2019-10-18T22:07:46Z

# start the command batch
batch start

board_name FF_RACEPIT_MINI
manufacturer_id FFPV

# name: AK

# feature
feature -RX_PARALLEL_PWM

# serial
serial 0 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1975 2025 0 0
aux 1 13 1 1975 2025 0 0
aux 2 26 0 1975 2025 0 0
aux 3 35 2 1975 2025 0 0
aux 4 40 3 1725 1825 0 0
aux 5 49 1 1475 1525 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_lowpass2_hz = 375
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 140
set dyn_notch_max_hz = 400
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_calibration = 41,23,60,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1025
set max_check = 2000
set rssi_channel = 12
set rc_interp_ch = RPY
set rc_smoothing_auto_smoothness = 5
set serialrx_provider = GHST
set blackbox_sample_rate = 1/8
set blackbox_device = NONE
set min_throttle = 1070
set dshot_idle_value = 650
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set bat_capacity = 1250
set vbat_max_cell_voltage = 445
set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 330
set vbat_scale = 111
set ibata_scale = 210
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1250
set osd_link_quality_pos = 12
set osd_tim_2_pos = 2455
set osd_throttle_pos = 2401
set osd_mah_drawn_pos = 2072
set osd_craft_name_pos = 2427
set osd_warnings_pos = 14377
set osd_avg_cell_voltage_pos = 2433
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_total_flights = ON
set osd_stat_total_time = ON
set debug_mode = RC_SMOOTHING_RATE
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5695
set vcd_video_system = NTSC
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 600
set stats_min_armed_time_s = 5
set stats_total_flights = 112
set stats_total_time_s = 6119
set name = AK

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dyn_lpf_dterm_curve_expo = 7
set dterm_lowpass2_hz = 180
set iterm_relax_cutoff = 30
set yaw_lowpass_hz = 100
set throttle_boost = 0
set p_pitch = 42
set i_pitch = 100
set d_pitch = 34
set f_pitch = 140
set p_roll = 41
set i_roll = 95
set d_roll = 34
set f_roll = 140
set i_yaw = 95
set f_yaw = 115
set d_min_roll = 0
set d_min_pitch = 0
set launch_control_mode = PITCHONLY
set ff_smooth_factor = 45

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 10
set roll_expo = 35
set pitch_expo = 35
set yaw_expo = 35
set roll_srate = 44
set pitch_srate = 44
set yaw_srate = 34
set tpa_rate = 40
set throttle_limit_type = SCALE

# end the command batch
batch end

# 
```

**Setup / Versions**
 - Flight controllers: RacePit Mini, Aikon F7 Mini, HGLRC Zeus F722 Mini
 - VTX: Tramp Nano, HGLRC Nano (Tramp protocol)

pgreenland, 2020-11-12:
Hey @akfpv,

Firstly, good shout on tagging me. I don't often contribute to open source projects for fear of introducing bugs 😬.

In making the first set of changes in PR #9638 I wasn't intending on changing any behaviour past improving the comms implementation and correcting the issue I'd identified while entering pitmode.

Having said that I've always been running my drone with pitmode tied to arming, I don't fly in any competitions so for me its just a thermal management thing, trying to keep the VTX cool while not flying.

Thats what PR #10166 covered, allowing the VTX to step out of pitmode once armed.

I've just run up your config on BF4.3, building the latest master and while I don't see your exact issue I do see something unexpected and likely incorrect.

As I power up, my VTX starts in normal mode, briefly enters pit mode and then returns to normal mode.

From inspecting the code I can see why this may happen. I've aired on the side of caution in the vtx driver, assuming a safe state for the vtx would be pitmode. It seems that my VTX (a Matek VTX-MINI) starts as expected in normal mode which it reports. As soon as the vtx driver contacts it it during initialisation it instructs it to change into pit mode, which it does. Somewhere around this time the driver starts advertising that it now has status information available for the vtx and the common vtx layer above the driver steps in, noticing that the current pitmode state doesn't match that set by the RC (or without the RC, normal mode) and requests it change, which it again does.

I'll look to correct that as it isn't all that nice, however it doesn't quite match what you're seeing. After the pattern of pit, run, pit, run you see does the unit become stable at the proper transmit power?

Are you able to switch in and out of pitmode using an RC mode with BF4.3 and this code as expected?

I'll prepare a fix for what I've spotted and drop a link on here to the branch....would you be in a position to build and test?

Thanks,

Phil

akfpv, 2020-11-12:
> Hey @akfpv,
>
> Firstly, good shout on tagging me. I don't often contribute to open source projects for fear of introducing bugs 😬.
>
> In making the first set of changes in PR #9638 I wasn't intending on [...]

Hey Phil,

Yes, the VTX seems to be stable once it settles. I don't use VTX pit mode, instead I use the FC pit switch (turning VTX on/off completely) function. But regardless of that, I don't think the default behaviour should have been changed in first place - there's a vast number of pilots who are well used to how it works right now and the new behaviour does not cater for everyone. I believe there already is a function that sets VTX power to low on disarm - maybe that's something you'd find useful for your particular scenario?

Could you please revert the logic to its original? Default logic should be VTX powering ON at the set power level on every plug in, unless majority of pilots/devs think otherwise.

Kind regards,
Alex

pgreenland, 2020-11-12:
Hey Alex,

Out of interest how long does the settling take for your vtx? Are we talking fractions of a second or seconds?

> But regardless of that, I don't think the default behaviour should have been changed in first place.

As I mentioned my intention wasn't to change the behaviour, you've identified a bug in the new code, I had to choose an initial state for a boolean variable and chose a safe one.

It's taken a while for someone to notice this given that it went in on the 26th of July in that first pull request, well spotted :-P.

Could you give the updates in my fork [vtx_irc_tramp_prevent_brief_pitmode_at_startup](https://github.com/pgreenland/betaflight/tree/vtx_irc_tramp_prevent_brief_pitmode_at_startup) a try and let me know if the issue has been resolved?

Thanks,

Phil

Thanks,

Phil

akfpv, 2020-11-12:
It takes 5-10 seconds I'd say (it varies), and also I had VTX turning into pit mod mid flight when rx was failsafing - so not fractions of seconds and there's definitely a safety concern here. I think @JyeSmith raised an issue #10158 sometime ago related to the same thing, but it got stale.

In case there are concerns about Tramp protocol implementation - I think @tonycake could help. Could you please explain what the original issue was that your PR was supposed to fix?

As for your branch I'll have a look at it tomorrow - would appreciate if you could make a F405 hex for me cos I don't have a build environment on this PC.

Thanks
Alex

JyeSmith, 2020-11-12:
@akfpv I lost interest in tramp out of frustration.  Tony is active here and can either fix it properly, or at a minimum release a protocol so devs have something to aim for.

akfpv, 2020-11-12:
> @akfpv I lost interest in tramp out of frustration. Tony is active here and can either fix it properly, or at a minimum release a protocol so devs have something to aim for.

Personally I don't know what needs, or needed, to be fixed - Tramp and Tramp compatible VTX's have been working flawlessly for me for the last 3 years (racing).

pgreenland, 2020-11-12:
Hi All,

Again thanks for spotting the issue, probably best both PRs are backed out.

I haven't got any more time to dedicate to this, as I said before I generally don't contribute to open source for lack of having the time to offer any support.

As for the issues I was trying to solve, they're detailed in my original PRs.

Good luck,

Phil

Igorshp, 2020-11-13:
I have a vtx (nlrc d400) that has a quirk when changing power levels.

When power level is changed, it briefly goes into lower power state, cutting video (unless i'm super close to it) for ~2-3 seconds. Naturally doing this in-flight is a bit uncomfortable.

I always assumed it was a quirk of this specific vtx as all my other ones don't show this behaviour. After reading this, I realised that all others use smartaudio.

Is it possible the behavior mentioned is related to this problem and is coming from betaflight?

Justicefpv, 2020-11-13:
Commenting to highlight this issue:

I have also noticed this issue but I am now unable to exit pit mode altogether (tramp nano) were as on previous builds prior to this 'fix' functionality has been perfect.

pgreenland, 2020-11-14:
Hi All,

Having spent some time debugging this with @akfpv we've made several discoveries.

PR #9638 will have fixed the tramp protocol for some users and broken it for others. It seems that some VTX units don't implement the original tramp protocol correctly. This PR therefore fixed it for my Matek unit which appears may not implement the protocol correctly and broke it for others.

PR #10166 added the ability to leave pitmode while arming a craft, however broke being able to properly control pit mode via the OSD or MSP.

Additionally I spotted an issue whereby the new tramp driver may briefly, incorrectly request that pitmode be entered on startup.

PR #10348 should address all three of these issues.

It adds a CLI option to support tramps which use inverted logic in their pitmode reporting, by default set to suppor the official protocol but allowing units like my matek to play nicely.

It backs out of PR #10166 until it can be re-implemented without introducing bugs.

It corrects the startup issue and no longer toggles pitmode on startup.

Please test with the unified builds attached to the PR and report back.

Ideally exercising as much tramp functionality as possible, power, freq and pitmode changes and provide the model of your VTX.

Thanks,

Phil

closed 2021-02-08 by github-actions[bot]