betaflight/betaflight #10346: There isn't Target in the Target list for a Matek f722 PX  fc

deltawhiskey1147, 2020-11-14:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

haslinghuis, 2020-11-14:
The target is `MATEKF722HD`
http://www.mateksys.com/?portfolio=f722-px

deltawhiskey1147, 2020-11-14:
Thank you very much

closed 2020-11-14 by deltawhiskey1147