betaflight/betaflight #10347: Failsafe Land mode alteration

thepolishcamera, 2020-11-14:
I had a recent mishap with a failsafe while flying a “cine-lifter”. It happened at a distance of ~150-200ft. So GPS Rescue would not be viable.

As I lamented the loss of a camera I started to wonder if the Land failsafe mode could be altered so it would not shut off the motors until it hit the ground. 

My current issue with how land works is this: If you’re higher than your duration is set for it may not matter that much. Because once the motors cut out gravity will do it’s thing and the heavier side of the quad will end up hitting the ground first. Usually being battery and camera side. As it did in my case. 

If it does it when you’re really low. Then it’s gonna hit the ground and the blades will chop up whatever is in its path. 

Forgive me for any naivety on how this actually works. 

My thought on how it might work is that when the quad is in “land” mode. It creates its own set-point (in a sense). And the gyro tracks with it. So what if the motors kept spinning until the set-point and gyro have a sudden disruption in how they track each other? E.g. hitting the ground.

  The hope here. Is that this sort of adjustment can help minimize the overall damage to drone/payload.

mikeller, 2020-11-14:
If properly configured the existing 'land' setting for failsafe will do exactly this - spin props at a throttle setting that is designed to reduce altitude, for a duration that is long enough to reach the ground at maximum altitude.
It sounds like this was not configured correctly on your quad.

Anyhow, adding a 'spin propellers until it hits the ground' function would be problematic for a number of reasons:
- if landing throttle is configured incorrectly and results in a hover / gain in altitude, this function will result in a 'fly towards the horizon' situation with a craft ever increasing altitude, until it finally drops from high altitude when the battery is exhausted;
- if landing throttle is configured correctly, but results in a slow decrease in altitude (which is desired), and the ground underneath is soft (say a bush), ground contact will never be registered, and a craft with a non-working RX link will stay on the ground with rotating props until the battery is exhausted.

thepolishcamera, 2020-11-15:
I can see the issues you presented. 

I didn’t think about someone setting the throttle value too high. 

I understand the challenge there. What if, instead of a set throttle value, it essentially went to “zero throttle” and just kept the quad level until impact? This may give the camera a better chance of survival?

I would think that the soft grass thing would still allow for the FC to realize it cannot correct itself to a level position and then shut off on its own.

haslinghuis, 2020-11-15:
You would need something like PX4Flow for precision landing.

thepolishcamera, 2020-11-15:
I was hoping to stay in the Betaflight ecosystem since FPV style videography is on the rise. 

I guess I was looking to see if amending the way the land feature currently works could help the survivability of the cameras mounted on “Cine-Lifter” style drones.

This would be before jumping into more complex accessories that might be able to aid in landings.

mikeller, 2020-11-22:
Hmm, there _might_ be a feature request in there:

We currently have support for sonar / LIDAR ground proximity sensing, but we do not use it for anything. One possible use would be for 'touch down detection' in 'land' and potentially 'GPS rescue' failsafe modes:

If enabled, while the craft is in phase 2 of failsafe 'land' mode, use 'altitude above ground' information from sonar / LIDAR to detect if the altitude is less than a set threshold, and if so, go into 'disarm' mode.

thepolishcamera, 2020-11-22:
Something like this would be amazing. LIDAR support would be great. Anything to help people with protecting their investments on these rigs would be amazing.

Kritik-Agrawal, 2021-05-31:
I've flown a large quad with an old Pixhawk clone running Ardupilot. While I do realize that Ardupilot/PX4 and Betaflight are completely different, with communities focused on different things (automated flight Vs high performance manual FPV flight), I think an inspiration can be taken from Ardupilot "Land" flight mode. I am not sure how it was implemented in Ardupilot, but the quad first got to the hover throttle in Auto-level mode(hover throttle learnt automatically in AP, but will have to be entered manually) and then it slowly reduced the throttle very gradually, with a controlled descent.

tavdog, 2022-08-31:
Is this being worked on at all?

haslinghuis, 2022-08-31:
@ctzsnooze is working on improving gps rescue and landing having return to home with landing. So there is active ongoing work for further improvement.

closed 2022-09-05 by haslinghuis

ctzsnooze, 2022-09-06:
Yes I think these issues are greatly improved in 4.3, and we have PR's for 4.4 to test that provide further improvements.