betaflight/betaflight #10351: Reboot / reboot into MSC does not work on H743

mikeller, 2020-11-16:
With latest master (e7edfa99e), rebooting the firmware, and rebooting into MSC does not work any more:

Reproduction for rebooting:

- enter CLI;
- type 'exit' [Enter].

Expected behaviour:

- the flight controller reboots;

Actual behaviour:

- the flight controller hangs and needs to be power cycled

Remarks:

When runing c6d70214f (before the merge of #10247) this issue does not occur.

Reprodoction for MSC mode:

- enter CLI;
- type 'msc' [Enter].

Expected behaviour:

- the flight controller reboots into MSC mode;

Actual behaviour:

- the flight controller reboots into DFU mode.

Configuration:

Bare IFLIGHTH743AIO flight controller with no peripherals attached.

```
# diff

# version
# Betaflight / IFLIGHT_H743_AIO (IF7A) 4.3.0 Nov 15 2020 / 17:35:42 (e7edfa99e) MSP API: 1.44

# start the command batch
batch start

board_name IFLIGHT_H743_AIO

profile 0

rateprofile 0

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: FREE
B01: FREE
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: ADC_RSSI
C06: FREE
C07: FREE
C08: CAMERA_CONTROL
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: LED 1
C14: FREE
C15: FREE
D00: GYRO_EXTI
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: LED_STRIP
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: FREE
E02: SPI_SCK 4
E03: FREE
E04: OSD_CS
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1: LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

SteveCEvans, 2020-11-18:
Fixed in https://github.com/betaflight/betaflight/pull/10352

SteveCEvans, 2021-02-05:
I believe this can be closed.

mikeller, 2021-02-09:
Yes, working now.

closed 2021-02-09 by mikeller