betaflight/betaflight #10353: [FEATURE REQUEST] Failsafe : "LAND" after failed sanity checks

2N5R, 2020-11-18:
Wouldn't it be a good idea to add an option that prevents a quadcopter from disarming (falling out of the sky) in case of a failsafe event with sanity checks failed? An option like "land in case of sanity check failed" would make sense.

Happend today: Failsafe event, GPS lost right in that moment, quad dropped on the concrete from 250ft height. A mode "land in case of sanity check failed" would have prevented the destruction.

Thanks and greetings

etracer65, 2020-11-18:
How would you suggest that the quad would "land" safely? In failsafe so completely uncontrolled. No GPS so it has lost any ability to determine it's position, heading, and likely altitude. So why would it be "safer" to have a completely uncontrolled blender potentially descending towards people or property?

closed 2020-11-18 by etracer65

2N5R, 2020-11-18:
Hi etracer,

if the quad disarms it just falls out of the air - uncontrolled and with lots of kinetic energy, depending on the altitude. It hits the ground and potentialy people, animals, cars, etc.
You are right - the quad has no idea where or how high it is. But if the quad apllied a predetermined RPM and activated self leveling mode, it would descend much slower and the impact would be much lighter. Still, it would hit approximately the same spot. But with much less kinetic energy. 

Does that make sense to you?

etracer65, 2020-11-18:
> Hi etracer,
> 
> if the quad disarms it just falls out of the air, is falls uncontrolled and has plenty of kinetic energy, depending on the altitude. It hits the ground and potentially people, animals, cars, etc.
> 
> You are right - the quad has no idea where or how high it is. But if the quad apllied a user-defined descend RPM and activated self levelling mode, it would descend much slower and the impact would be much lighter. Still, it would hit approximately the same spot, but with much less kinetic energy. That could protect the quad and people/animals/objects it hits.
> 
> Does that make sense to you?

No, it does not. First of all it's your responsibility to NOT fly in an unsafe manner over people or property. Secondly, having an uncontrolled machine spinning dangerous propellors at high RPM looking to shred anything in it's path is in no possible estimation "safer". Also consider what might happen if this uncontrolled quad crashed into some object or the ground while it was still armed and trying to spin the motors. In the best case it will burn the motors and ESCs, in the worst it's possible it will start a fire in the process.

mikeller, 2020-11-22:
All of the above safety concerns are ones that equally apply to the 'Land' failsafe mode - this is an accepted risk, and means that this mode can only be used in situation where the risk is appropriately mitigated - i.e. when **exclusively** flying in areas where there is nothing on the area that is at risk of damage / fire.
So I think as a possible way to make the configuration a bit more flexible, it should be possible to select one of the existing failsafe modes as the 'GPS rescue fallback mode' - this does not introduce risks that aren't already there.

2N5R, 2020-11-22:
mikeller, that is what I mean. Thank you!