betaflight/betaflight #10358: enhanced vtx_info like feature ?

pitts-mo, 2020-11-20:
I would like to provide users with improved awareness of VTX telemetry health and VTX features. I currently believe it may be desirable to create an enhanced variant of the vtx_info command.

Ideally this new feature/command would report user readable information of :
1 - general status of configured VTX as responding / not responding
2 - hardware, firmware, telemetry version if available
3 - basic features and power levels if available

However I do recognize it would likely be simpler if this command just runs through a list of known commands to query the configured VTX and just return "raw" data into the CLI.

This idea has came from a recent hour plus I consumed while assisting a friend with a 3rd party smart audio device. And was further pushed by a recent Joshua Bardwell / Jon E5 FPV video.
https://youtu.be/ag09Rae0P_g?t=879

Thoughts?
Thank you,
-p

some of my related thoughts:
smartaudio debug is a bit too cumbersome 
vtx_info is a little helpful

JyeSmith, 2020-11-23:
A lot of this info is provided by the VTx and BF could use it to auto setup most of the VTx table e.g. Tramp and SA already reply with the allowed power levels and we know what freqs they use, disregarding locked VTxs.

The bigger issue will be 3rd parties not implementing the protocols correctly.  So now any auto setup or useful user information may be missing.

I started (slowly) working on JyeSmith/OpenVTx with the plan to make it easily flashable onto any stm8 VTx.  The plan is to have SA AND Tramp usable on the VTx, and to make the protocols more flexible e.g. SA should allow any power level up to max like Tramp.  3rd parties would be free to pick it up and hopefully that'll make life easier.

pitts-mo, 2020-11-24:
Ok,
I will start to slowly work on "vtx_status" as well. 

I had noticed a lot of VTX query routines already exist in BF. My current plan is do start by duplicating vtx_info as vtx_status and use the existing routing to indicate the configured VTX type and then conditionally call existing routines for the devices status details.

Also noticed vtx temperature should be included.

Thank you for the thoughts,
-p

pitts-mo, 2020-11-26:
I have noticed the cms_menu_vtx_* files contain most of what is needed already... I figure these could be copied, converted for cli specific display, and then enhanced for the vtx_status feature.

@mikeller can you please provide guidance?
For cli/cli.c... should I attempt to:
a- incorporate all the logic and cliPrintLinef tasks for the various VTX directly into the cli/cli.c?
b- fork out a majority of the logic and/or cliPrintLinef tasks into separate files?

bonus thoughts:
My Initial testing has also enlightened me to a few new aspects of user confusion within the diversity of configuring a VTX. The most interesting new item is multiple VTX configured... However, searching for every possibly configured VTX type, it's connection method, and labeling it or processing it as valid for control seem excessive at this time.

mikeller, 2020-11-26:
@pitts-mo: My hunch would be to try and avoid duplication - move any complex functions that return VTX info into a common place like `src/main/io/vtx.c`, and then use them to build output in CMS / CLI.

pitts-mo, 2020-12-08:
I have not found much time to look at this. However, I want to add some notes and observations here in case I (or someone else) have time to pick this up again.

In short I agree with @mikeller.

task #1. unify existing VTX related OSD menus:
Since the VTX OSD menus already do most of the work and appear to be working acceptably... It currently seems that all cms/cms_menu_vtx_* items should be unified for displaying status and configuration of all VTX types from a single/standardized OSD menu. All query and configuration code for specific VTX would need to be moved into their corresponding io/vtx_* files. This would require standardization of the query routines and the data returned as much as possible. But we could always keep a couple generic status queries that allow displaying information unique to specific VTX (I remember seeing something like this already existed for at least one VTX type).

task #2. add "vtx_status" command to cli:
With #1 complete it should then be much easier to add a "vtx_status" command into cli. 

Finally thought: 
The changes suggested here should make it a little easier to add complete support for additional VTX types or VTX related interfaces going forward.

-p

limonspb, 2020-12-27:
Something similar?
https://github.com/betaflight/betaflight-configurator/pull/2349

closed 2026-09-26 by blckmn