betaflight/betaflight #10361: External barometer not visible

maxgame91, 2020-11-22:
hello, just bought an external barometer [here](https://www.amazon.fr/HiLetgo-BMP280-3-3-Precision-Atmospheric-Navigation/dp/B07ZWV68M6/ref=sr_1_6?__mk_fr_FR=ÅMÅŽÕÑ&dchild=1&keywords=BMP280&qid=1606070229&sr=8-6), and betaflight do not show on the top right section (it's activate in the configuration menu.
So it's a BME280

i have an F4 Mamba so the target is FURYF4OSD
I solder the SDA to the SDA and the SCL to the SCL
On betaflight 4.2.4
No I2C errors. on the status command.

Do you have any idea why

If you need any more information just tell me.

Thanks

mikeller, 2020-11-22:
@maxgame91: Yes. Diatone does not want to support barometers on FURYF4OSD - buy yourself a different flight controller.

closed 2020-11-22 by mikeller

etracer65, 2020-11-23:
@maxgame91 Assuming you connected it properly to the SCL and SDA pads and supplied power to the board, paste the following into the CLI and it should work:
```
set baro_hardware = AUTO
set baro_bustype = I2C
set baro_i2c_device = 1
save
```
But note that the device you linked to appears to require a 3.3V supply so you don't want to power it from 5V. And as far as I can tell the Diatone Mamba F4 board doesn't have a 3.3V output. If you power it from 5V you'll damage the sensor and possibly your flight controller.

mikeller, 2020-11-23:
As stated above, doing this modification is not supported by Diatone and will void any warranty.

etracer65, 2020-11-23:
> As stated above, doing this modification is not supported by Diatone and will void any warranty.

LOL! This is obviously completely untrue. The unified target just has errors and Diatone had nothing to do with that. The legacy target for FURYF4OSD is correct and does support an external baro.

mikeller, 2020-11-23:
Whatever @etracer65, it is ok if you want to disregard the manufacturer's explicit wishes, but please stop spreading bullshit and misinformation to unsuspecting users.

etracer65, 2020-11-23:
@mikeller Could you please link us to some statement from Diatone that "connecting a Baro" will void any warranty?

Further the unified target with the errors was not created or submitted by Diatone. And as I already stated, the legacy target does support both the baro and mag connected to I2C using the pads provided by Diatone on the board.

I realize you like to use the tactic of shaming manufacturers when they submit unified targets with the mag or baro disabled by threatening to tell their customers to "buy another board if they want baro support", but in this case Diatone had nothing to do with it. You in fact created the incorrect unified target.

mikeller, 2020-11-23:
@etracer65:

> Could you please link us to some statement from Diatone that "connecting a Baro" will void any warranty?

With pleasure:

https://github.com/betaflight/unified-targets/blob/8d68f6c8ed65e366d05ac59d9a49b10912ac3301/configs/default/DIAT-FURYF4OSD.config#L87

The manufacturer or their target maintainer (@kc10kevin) clearly do not want barometers to be supported on their hardware, or else they would leave this setting on default (`AUTO`).

> You in fact created the incorrect unified target.

Please stop spreading lies in here - this will not be tolerated.

etracer65, 2020-11-23:
> 
> > You in fact created the incorrect unified target.
> 
> Please stop spreading lies in here - this will not be tolerated.

Is the truth tolerated? https://github.com/betaflight/unified-targets/pull/9

So blaming the manufacturer is not really a viable argument. But regardless it really shouldn't be about "blame" and thus hurting the users. Rather than suppress information wouldn't it make more sense to correct the errors so that other users don't run into this problem in the future? Or is it more important to punish the manufacturers?

mikeller, 2020-11-23:
@etracer65: Of course the truth is tolerated:

[image]

I just wish it was the full truth:

[image]

> So blaming the manufacturer is not really a viable argument. But regardless it really shouldn't be about "blame" and thus hurting the users. Rather than suppress information wouldn't it make more [...]

This is where I disagree.
Betaflight currently supports ~210 targets, with new ones added on a weekly basis, and the quality of new ones often being poor, with 'target maintainers' who barely manage to follow the instructions.
To me this is the area where manufacturers need to take responsibility and own the targets for their hardware.
Decisions as to whether a manufacturer wants to support barometers or not (some explicitly said they don't, after discussion) need to originate from the manufacturer, and having volunteers provide what is perceived as a fix (but could just as well be against the manufacturer's intention) is not a good approach.

As an alternative way forward, we could try pinning issues like this one onto the target maintainer, and allow them to provide the correct solution (either 'not supported' or a fix for the target). Moving this.