betaflight/betaflight #10364: MSP do not have "link_quality_info" CMD

ligenxxxx, 2020-11-23:


etracer65, 2020-11-24:
There has never been a need to display link quality in the Configurator so therefore no need for a MSP message related to this. Link quality is rarely needed on the bench when connected to the Configurator.

mikeller, 2020-11-24:
@ligenxxxx: May I ask what you are trying to achieve that would need to have a command to read link quality info over MSP? Most likely there is a better way for you to get the desired result.

ligenxxxx, 2020-11-28:
At present, all digital video transmission red copper only obtains osd information through msp. If all osd information can be obtained through msp, then digital image transmission can support all osd

mikeller, 2020-11-28:
@ligenxxxx:

> At present, all digital video transmission system only obtains osd information through msp. If all osd information can be obtained through msp, then digital image transmission can support all osd

Using MSP query commands to poll information for an OSD is not a scalable or good approach. This will only ever work well for OSD applications with a limited feature range, and will make it impossible for users to use the graphical configurator of the Betaflight OSD that they are used to.
If you want to support full OSD capabilities on your device you should be using displayport over MSP.

ligenxxxx, 2020-11-30:
ok,thanks

closed 2020-11-30 by ligenxxxx