betaflight/betaflight #10366: GPS performance difference F7 <-> F4

DeadT0ast, 2020-11-23:
**Describe the bug**
I have bought a Flywoo Explorer LR analog edition and thought the GPS functionality was a neat feature. So I decided to add a GPS module to my 5inch quad too just in case I would loose video and for some logging purposes. The GPS is mounted in the same way it is on the Explorer LR. Yes I am aware that it is not an optimal position for the module but the location is not the cause for this issue (I have placed my quads back to back with the Flywoo GPS sitting on top of the 5inch GPS just to rule out the location causing the issue).
Now the actual issue is that I get really poor performance from the GPS mounted on my 5 inch quad with an iFlight SucceX F7(EXF722DUAL) FC. So first I thought the TBS M8 Module was to blame and I swapped the Module with the Flywoo one from my Explorer LR. But I still have poor performance on the 5 inch quad while the TBS M8 works just fine on the Explorer LR with a FlywooF411 FC.
When checking u-center with the "gpspassthru" CLI command I realized that whatever GPS module is mounted on the F7 board gets no Galileo satellites, while the one mounted on the F4 does. 
This makes me believe that the different FCs send different GPS setup commands to the ublox modules.
I have tested these things with only USB connected and no battery. So only the FC and the GPS are powered (and the CF nanoRx on the Flywoo). Doing the same tests with battery does not change the results.

Poor performance:  satellites < 6 and dropping to 0 satellites during flight, GPS rescue not possible
Working fine: satellites > 15 and sometimes dropping to < 8 sats during flight, GPS rescue 95% of the time possible 

**To Reproduce**
Steps to reproduce the behavior:
Compare GPS performance (with u-center) with any GPS on a F7 vs F4. MY F7 gets no Galileo sats while the same module on a F4 does.

**Expected behavior**
The GPS modules should not behave differently on a F4 vs F7 (at least not F7 worse than the F4)

**Flight controller configuration**
Flywoo F411:
```
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.2.5 Nov 22 2020 / 18:38:45 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: FLWO, board_name: FLYWOOF411, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FLYWOOF411
manufacturer_id FLWO
mcu_id 0058003e3030510435393433
signature 

# name: FLYWOO

# resources
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A15

# timer
timer A01 AF1
# pin A01: TIM2 CH2 (AF1)

# dma
dma pin A01 0
# pin A01: DMA1 Stream 6 Channel 3

# mixer
mmix 0  1.000 -0.809  0.659 -1.000
mmix 1  1.000 -1.000 -0.659  1.000
mmix 2  1.000  0.809  0.659  1.000
mmix 3  1.000  1.000 -0.659 -1.000

# feature
feature -RX_PARALLEL_PWM
feature -LED_STRIP
feature GPS

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
[... 413 lines cut ...]
set yaw_rc_rate = 120
set roll_srate = 64
set pitch_srate = 64
set yaw_srate = 72
set tpa_rate = 55
set tpa_breakpoint = 1400

# end the command batch
batch end

# 
```

**Setup / Versions**
 - iFlight F7(EXF722DUAL) and FLYWOOF411
 - TBS M8 GPS module and FLYWOO GM8 MINI V1.0 GPS
 - Both GPS modules wired over a hardware UART

**Additional context**
Additional screenshot from u-center. Both quads sitting next to each other outside my window so only the western side of the sky is visible to the Modules. 
u-center Flywoo with TBS M8 GPS
[image]

u-center iFlight F7 with Flywoo GPS
[image]

mikeller, 2020-11-23:
Other hardware like VTX, cam, ESCs can all interfere with GPS reception - have you checked if only powering the flight controller and GPS gives you a better result? If so, add one component at a time to find out which one results in a drop-off.

DeadT0ast, 2020-11-23:
I have it only powered with USB and no other components. Running the quad with all components on doesn't change anything significant.
Anyways that would not explain why only the F4 board is able to see Galileo sats and the F7 not

DeadT0ast, 2020-11-23:
I have just soldered one Module to a SpeedybeeF7 and I do get Galileo sats so my assumption of it being a F4<->F7 problem is wrong. But there seems to be a difference on my iFLight FC since no matter what it will not see any Galileo sats. I have strapped my second GPS module connected to the SpeedybeeF7 on top of the one connected to my iFlight FC and it still sees Galileo sats. Both FCs were running at the same time so I don't think the iFlight FC caues interference to the GPS module. (Again all the tests with only the FC and GPS powered, no other components)

DeadT0ast, 2020-11-23:
Did another test with a F4 from my Beta95x v2 and it also detects Galileo. 
When I take a module that already had a fix including Galileo sats and I connect it to my iFLight FC it keeps the Galileo sats until it does the next coldstart. But then again it won't even see one Galileo sat.

mikeller, 2020-11-23:
@DeadT0ast: This is odd. This is above my level of expertise in GPS / Ublox configuration, but could it be that the configuration pushed by Betaflight just results in Galileo satellites being picked up extremely slowly?

DeadT0ast, 2020-11-24:
Yes it is really odd. Especially since it usually picks up Galileo sats in a reasonable time with any other combination except that iFlight FC. But it also doesn't even show a singe Galileo sat. Usually in u-center it starts showing a bunch of satellites it sees or has at least picked up for a microsecond or so and there are a bunch of GPS, GLONASS and two or three EGNOS sats. But not a single glimpse of a Galileo sat.
Is it even possible that the iFlight FC sends any different Ublox configuration packets than the other FCs?

DeadT0ast, 2020-11-24:
Just tried the same thing with INAV and it shows the exact same behaviour for that FC. I don't know if that helps anything or how much the GPS code is still related to eachother.

McGiverGim, 2020-11-24:
I suppose that yes, but the GPS module is the same for both FC?

DeadT0ast, 2020-11-24:
I have one TBS M8 module and one FLYWOO GM8 MINI V1.0 GPS. I have switched them both back and forth but it does not matter which one is connected. They both behave the same. Both work on all my F4s and the Speedybee F7 and both modules show no Galileo on my iFlight.

etracer65, 2020-11-24:
> Is it even possible that the iFlight FC sends any different Ublox configuration packets than the other FCs?

Absolutely not.

Not really sure what the big mystery is. All electronic devices emit RF. Your "problem" flight controller is simply emitting RF that's interfering with the frequencies used by the satellite constellation. The iFlight board clearly has some design flaws and is simply "noisy". You'e proven this yourself by showing no problems with other flight controllers and the same behavior with different firmwares. Just accept that the iFlight board is not a good choice for use with GPS.

DeadT0ast, 2020-11-24:
I didn't really think it would get any other config. I do understand that the iFlight FC might be especially noisy. But I can tape the other gps module on top of the one mounted in the iFlight (and of course powering the iFlight FC on) and it gets sats just as quick as it does far away from the iFlight FC. Also why would it then still see the Galileo sats when I swap the module over from any other FC before it looses the almanac or whatever? I would think that it would loose them because of the RF noise that is somewhat only interfering with Galileo sats. I'm not trying to defend this FC. It's just the one I am using now and I would have liked to get it working on this one.

etracer65, 2020-11-24:
The "noise" doesn't have to be radiated and could also be coming from the power being supplied from the iFlight controller. A noisy power supply could affect the GPS module's circuitry. But all this is really going way beyond the scope of appropriate conversation here. There is no mechanism in Betaflight to configure the GPS module differently based on flight controller architecture or brand. Further you've eliminated the firmware from the picture by experiencing the same problem on INAV. We don't have any knowledge or details about how the GPS module internally manages connections to different satellite constellations. It's clear that this problem you're having is related to the electronics in the iFlight flight controller. I would recommend contacting iFlight for support, or simply not using that flight controller with a GPS module.

DeadT0ast, 2020-11-24:
Alright. I hoped it could be anything else. I am aware that these modules are very sensitive to noise. I just expected them to like have general poor performance and not just having one group of satellites not show up at all. Anyways I guess I will be looking for another FC in the future and keep this one as backup.

closed 2020-11-24 by DeadT0ast