betaflight/betaflight #10371: Betaflight FC as gimbal controller

stackola, 2020-11-27:
Quadcopters flight controllers and gimbal controllers aren't all the different, they utilize the same type of sensors and use a PID loop to control some brushless motors. That made me wonder how hard it would be to misuse betaflight along with a random flight controller as a gimbal controller.

**Describe the solution you'd like**
I would like Betaflight to come with an additional mixer for 1,2 or 3 axis brushless gimbals. The flight controller would be mounted directly to the camera, and not have any part in controlling the quadcopter it self. The PID loop could probably be re-used for that purpose.

**Describe alternatives you've considered**
I've been looking for open source alternatives but could not find anything that would support standard flight controller hardware. The advantage of having this included in betaflight is having all the board definitions, sensor support and great PID controller already included.

**Additional context**
Additionally, the gimbal controller could accept PWM signals from the flight controller to accomplish things like CAMERA_TILT.

etracer65, 2020-11-28:
Seems more appropriate as a fork dedicated to gimbal control as only a relatively small part of the existing Betaflight code would be involved. Also since camera platforms with gimbals are not really part of the Betaflight project's goals it would be undesirable to include code for this in the main project when we could use the limited code space for things related to flight performance. The required code would be significant and only a tiny number of the typical Betaflight users would benefit.

mikeller, 2020-11-28:
It has to be said that there probably isn't much in Betaflight that can be used for a gimbal controller, besides 'read gyro / acc' and 'output PWM' (assuming that the servos you want to use for they gymbal even use PWM input). Servo output control in Betaflight is rather crude, and I doubt that it is good enough for a well working camera gimbal controller.

stackola, 2020-11-28:
Thank you for your opinions!

I'll think about the fork-angle, and close this issue.

closed 2020-11-28 by stackola