betaflight/betaflight #10374: send RPM telemetry when Bidirectional Dshot is enabled

ubanov, 2020-11-29:
If I want to have ESC telemetry currently the only way to do it is use one additional serial cable, and enable MOTOR ESC. I think few people do it (all of us allways need more and more serial ports).

Right now if I enable Bidirectional dshot, (I think that this feature has a lot of potential to become a very used feature), FC has the RPM information, but it doesn't send it to the receiver (in the current release, I can see it in section motors, of the BF Config App, but not in the receiver, and I would to see it in fly, not in test mode).

The solution I'd like:
If Motor ESC is not enabled, and the FC has Bidirectional Dshot enabled, send this RPM information each 1/20-1/50 of second to the receiver.

mikeller, 2020-11-29:
@ubanov: Currently there is no easy way to do this, as there is no room in the Dshot feedback frame to indicate if the frame contains RPM or if it contains 'other' information. And the length of the frame can not easily be extended, as it is pushing the limits of what is feasible on BLHeli_S.

ubanov, 2020-12-01:
I don't know how the code is, but... if I enable bidirectional dshot, the FC it's already receiving RPM information. Then this process should not be modified... the only thing that would be neccesary is to store RPM information for the telemetry process. wouldn't it?

mikeller, 2020-12-01:
@ubanov: Hmm, I guess I should be asking differently: *Where* do you want to see / use the RPM information?

ubanov, 2020-12-02:
I want to be able to see rpm information when drone is flying... Then what I have proposed is to send the rpm information using telemetry (receiver . Currently is possible to have this information, but need to enable MOTOR ESC (and connect one serial cable between ESC-FC). What I proposed is that the source for RPM telemetry could be changed, if the FC already have it.

mikeller, 2020-12-02:
@ubanov: Ah, so what you are proposing is to have the RPM information sent to the TX over telemetry? Yes , this could definitely be done. In most cases doing this will be up to the manufacturer of the RX / TX system, as they control what fields can be sent over telemetry, and not Betaflight.
So unless you can specify what RX / TX system / protocol you want to use there is not much we can do.

etracer65, 2020-12-02:
@ubanov You can already display the motor RPM in the OSD using the ESC RPM element. If you're not flying FPV I'm not sure the value of having the RPM in the radio telemetry. Hopefully if flying LOS you aren't looking down at your radio to watch the motor RPM. Betaflight doesn't have autonomous flight modes so there's not a situation where it's safe to take your attention away from flying. And yes I'm aware that the ESC sensor RPM data is currently sent through smartport telemetry but that's perhaps a bad decision that should be revisited.

closed 2021-02-15 by github-actions[bot]