betaflight/betaflight #10375: Dshot commands, disableMotor() and Bidirectional dshot not working together

limonspb, 2020-11-30:
Blocking dshot commands (with disableMotor()) are not working if bidirectional dshot is enabled causing FC to freeze.

**To Reproduce**
0. Take props OFF.
1. Turn on bidirectional dshot, save the configuration.
2. Plug the battery.
3. Open CLI
4. type "DshotProg 0 1"

**Expected behavior**
Motor "0" should make a short beep sound.

**Flight controller configuration**
Any FC

**Setup / Versions**
Current master branch

haslinghuis, 2020-11-30:
Seems to be working on my latest fixes #2290.

haslinghuis, 2020-11-30:
Seems to be working on my latest fixes https://github.com/betaflight/betaflight-configurator/pull/2290

mikeller, 2020-11-30:
Thanks for testing @haslinghuis - good to know it seems to be MCU specific. @limonspb: Can you please supply what flight controller / configuration was used to reproduce?

limonspb, 2020-11-30:
@mikeller 
I tested with Clracing F7.
Asizon tested on F7x2 and we both have this problem
I haven't tried F4 yet, will doo soon

limonspb, 2020-11-30:
Tried FuryF4OSD and the bug is still there.
Both 4.2 release and 4.3 master branches on default configuration (applied custom defaults)

I also applied DIFF from @haslinghuis and it did not help, the bug still appears if bidirectional is ON

limonspb, 2020-11-30:
Hm.... tried also on IFLIGHT_F411_PRO(STM32F**411**)
and the bug is NOT there.

mikeller, 2020-11-30:
Hmm, quick thought: What is the setting of `dshot_bitbang`, everybody. And please supply `timer show` as well if the setting is `AUTO`.

haslinghuis, 2020-11-30:
No I have tested with F411 (Matek) and a F446 and hear a short beep sound on both. Tested on Fedora 32 with master branch

limonspb, 2020-11-30:
@mikeller seems like the bug reproduces only when dshot_bitbang = AUTO. Here are my tests on F405 and F411
[image]

haslinghuis, 2020-11-30:
*Matek F411*
``` 
# get dshot_bitbang
dshot_bitbang = OFF
Allowed values: OFF, ON, AUTO

dshot_bitbang_timer = AUTO
Allowed values: AUTO, TIM1, TIM8
```
```
# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
```

haslinghuis, 2020-11-30:
Also applies for `dshot_bitbang = ON`

mikeller, 2020-11-30:
`AUTO` means 'it's on or off depending on the hardware configuration', so 'reproduces with `AUTO`' is not much help. :wink: 
But in this case you can see the actual status from `timer show`: If it contains lines with `DSHOT_BITBANG`  then it's on.

haslinghuis, 2020-12-05:
Trying to isolate the problem. Did some testing using 10.7 / 10.8 configurator using 4.2.0 / 4.3 firmware. After setting `dshot_bitbang = ON` and then using the `dshotprog 0 1` command stops cli communication completely. Issuing commands afterwards are ignored. Using several other cli commands before using `dshotprog` command have no issue.

haslinghuis, 2020-12-05:
Tested changing line 3863
https://github.com/betaflight/betaflight/blob/c9dd69b19e4a7c1e5bac2a17f4aef9ec170878d5/src/main/cli/cli.c#L3863
to `DSHOT_CMD_TYPE_INLINE`
With this console output does not stop outputting commands after using the `dshotprog` command. However the beep is not there.

limonspb, 2020-12-30:
Fixed here:
https://github.com/betaflight/betaflight/pull/10406

closed 2020-12-30 by limonspb