betaflight/betaflight #10379: Resource mapping not allowing for customized airplane with two motors and two servos

BigBernie23, 2020-12-02:
Latest Betaflight firmware releases (tried stable and development) on an Omnibs F4 nano V7.  Trying to setup a fixed wing with two motors (ideally differential thrust) and two aileron servos.   I can get to work either the motors or the servos, but can't find a way to have the four of them working at once.

Install Firmware, set Custom Airplane.   go to Cli and set up as follows (extracted from diff)

With this setup the motors work but the servos don't.   If I map the servos as servo 1 and servo 2 (instead of 3 and 4) then the servos work but the motors don't

**Flight controller configuration**
```
batch start

board_name OMNIBUSF4NANOV7
manufacturer_id AIRB

# resources
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource SERVO 3 C09
resource SERVO 4 B07

# mixer
mixer CUSTOMAIRPLANE
mmix 0  1.000  0.000  0.000  0.300
mmix 1  1.000  0.000  0.000 -0.300

# servo mixer
smix 0 3 1 100 0 0 100 0
smix 1 4 1 100 0 0 100 0
smix 2 3 0 100 0 0 100 0
smix 3 4 0 -100 0 0 100 0

# feature
feature -OSD
feature -DYNAMIC_FILTER

# map
map ATER1234

# master
set acc_calibration = 11,9,77,1
set mag_hardware = NONE
set baro_hardware = NONE
set motor_pwm_protocol = PWM
set pid_process_denom = 16

profile 0

rateprofile 0

# end the command batch
batch end

```

**Setup / Versions**
 Ominbus F4 nano V7 with 4.2.5 firmware

BigBernie23, 2020-12-02:
Intersting.  If I type resource I get this....
resource BEEPER 1 C05
resource MOTOR 1 C08
resource MOTOR 2 B06
resource SERVO 3 C09
resource SERVO 4 B07
resource LED_STRIP 1 A15
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 B12
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ADC_BATT 1 C00
resource ADC_CURR 1 C01
resource FLASH_CS 1 A02
resource OSD_CS 1 C15
resource GYRO_CS 1 D02
resource GYRO_CS 2 C04

with motors and servos correctly mapped

but if I ask for resource show all I get this

A00: FREE
A01: FREE
A02: FLASH_CS
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: LED 1
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: PREINIT 2
C05: BEEPER
C06: FREE
C07: FREE
C08: MOTOR 1
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: PREINIT 4
D00: FREE
D01: FREE
D02: GYRO_CS 1

where the servos are not mapped to C09 and B07 as intended.   Why the difference?

etracer65, 2020-12-02:
I think you want
```
resource SERVO 1 C09
resource SERVO 2 B07
```
not servo 3/4. Same in your `smix` settings.

BigBernie23, 2020-12-02:
You're absolutely right, that was my first attempt.  But when I do that I get the servos to work, but the motors won't.  Also if I map the resources this way still the smix has to reference to servos 2 and 3 to work.  If I reference to 0 and 1 it doesn't.

Thanks so much for your help

etracer65, 2020-12-02:
I tested with this layout (note servos 1/2 in the resource and 0/1 in the `smix`):
```
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource SERVO 1 C09
resource SERVO 2 B07

# mixer
mixer CUSTOMAIRPLANE
mmix 0  1.000  0.000  0.000  0.300
mmix 1  1.000  0.000  0.000 -0.300

# servo mixer
smix 0 0 1 100 0 0 100 0
smix 1 1 1 100 0 0 100 0
smix 2 0 0 100 0 0 100 0
smix 3 1 0 -100 0 0 100 0
```
and it appears to work fine. The motors/servos in the Motors tab seem to be behaving as expected and and respond to the controls and gyro/acc properly.

Test this yourself in the Motors tab to see if logically it's behaving as desired. If the motors still don't actually spin then it's likely a timer conflict between the motor pins and servo pins. Remember that before they were all motor pins. Also PWM motor output protocol is probably poorly supported anymore and will force your PID loop to 500hz so not desirable. If your motor ESCs can support it you would be better off with a more modern protocol - at least ONESHOT125. This probably won't resolve the timer conflict though.

I'm not really an expert on timers but it might work better if you use motor pads 1/3 (C08/C09) for the motors since they are  using timer 8 and motor pads 2/4 (B06/B07) for the servos as they are timer 4.

BigBernie23, 2020-12-02:
I just tested your first suggestion.  Unfortunatele nothing worked, no servos no motors.  The funny thing is that if I use this servo mix
smix 0 3 1 100 0 0 100 0
smix 1 2 1 100 0 0 100 0
smix 2 3 0 100 0 0 100 0
smix 3 2 0 -100 0 0 100 0
mapping servos 2 and 3 (even when the resource pins are mapped to servos 1 & 2) does the trick!  I see on the motors tab that the servos 3 and 4 are responding to the receiver and the servos actually moved.   Is there any other mapping underneath that servo 1 ends up being servo 3??  doesn't make sense but that works.

But still, no motors.   I like your last idea about the timers.  Let's try

Thanks!

BigBernie23, 2020-12-02:
IT WORKS!!  you were right, need to map both motors to C pins and both servos to B pins and it works
resource MOTOR 1 C08
resource MOTOR 2 C09
resource SERVO 1 B06
resource SERVO 2 B07

for some reason the smix still needs to refer to servos 2 and 3 in order to function but who cares, this works!

Thanks so much man!

jflyper, 2020-12-03:
@BigBernie23 
> for some reason the smix still needs to refer to servos 2 and 3 in order to function

I'm not following the conversation at all, but there is a chance that you are mixing up servo slots with logical servo numbers.

This is the servo demystifying chart I made few years ago, and should help you understand what slots are.

[image]

mikeller, 2020-12-06:
@BigBernie23:

> for some reason the smix still needs to refer to servos 2 and 3 in order to function but who cares, this works!

That's exactly this part in @jflyper's diagram:

[image]

`smix` refers to _Servo Slots_, `resource` refers to _Logical Servos_, the mapping between them is done by Betaflight in the order shown in the diagram.

closed 2021-02-15 by github-actions[bot]