betaflight/betaflight #10380: No telemtry altitude with barometer

maxgame91, 2020-12-02:
Hello,

So I have to build : 
- the fisrt one with a gps and baro on the FC
- the second with only a baro on the FC

When I search for new sensors wit the first build I have the telemetry for the altitude, but with the second build I don't have any altitude telemetry

So I guess this is how betaflight work ? Only the altitude calculate by the gps can be use for telemtry
Why the height define by a barometer are not use for the telemetry ? It's the only purpose for a barometer and it's far more precise.

Thanks

maxgame91, 2020-12-02:
@mikeller add betaflight version and hardware.

maxgame91, 2020-12-02:
@etracer65 add some info. If you need more things please tell me why. thanks

etracer65, 2020-12-02:
Read the info when you created the issue.

> **Describe the bug**
> A clear and concise description of what the bug is.
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> **Expected behavior**
> A clear and concise description of what you expected to happen.
> 
> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
> 
> **Setup / Versions**
>  - Flight controller [what type is it, where was it bought from];
>  - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
>  - how are the different components wired up.
> 
> **Additional context**
> Add any other context about the problem here.
>

maxgame91, 2021-01-13:
So I update with all the information ask, thanks

wildjokerss, 2021-02-11:
If it helps with ruling out anything, I'm having the same issue on Betaflight 4.2.6 in a Matek F405-CTR AIO. Crossfire 4.10 on a TX16S
Baro is working fine on OSD and also see the graph moving on the betaflight configurator. Radio has openTX 2.3.10.

If you need my help let me know and I'll get you more data, was troubleshooting this just now and landed in this bug report.

mikeller, 2021-02-13:
@maxgame91: Unfortunately in CRSF telemetry, the only place where the altitude is sent is in the GPS frame: 

https://github.com/betaflight/betaflight/blob/21eea5db21c88f927d6d955452734539ddfb4ec8/src/main/telemetry/crsf.c#L157-L178

To make matters more complicated, the altitude sent is _absolute_ altitude, which isn't available from a barometer only.

Not sure there is a good way to improve this unless a different frame type with a field for relative altitude is added to the protocol.

wildjokerss, 2021-02-19:
Question from someone that does not really understand much of programming. Isn't there a way to implement into Betaflight a CLI command to change between altitude sensors? As in, report barometer altitude like if it was GPS data? this would probably be a suitable solution. Is there another alternative? Like log a bug on TBS CRSF?

etracer65, 2021-02-19:
> Question from someone that does not really understand much of programming. Isn't there a way to implement into Betaflight a CLI command to change between altitude sensors? As in, report barometer [...]

This is not the problem. The problem is that CRSF telemetry only has as altitude field in the frame that contains GPS data. So the only way to send altitude is by sending all the GPS data. With no GPS we don't have anything to send.

mikeller, 2021-02-21:
@wildjokerss: The only way to implement a 'fix' for this in Betaflight will be to, in the case of a barometer but no GPS being found, GPS frames with bogus GPS location / speed / ... and a relative altitude in the field designated for the absolute altitude are sent. This would be a truly poor 'fix'.

wildjokerss, 2021-02-22:
Ok understood. So if I wanted to have Barometer telemetry on the radio, what would have to be done? Meaning, is it on betaflight side, crossfire side... Guess a feature request would have to be made to one of the two? For example, would it work with a telemetry enabled FrSky receiver?

Also, on the page 45 from the PDF below it states that the data source for Alt is TBS GPS or FC, I am guessing that it's either TBS GPS connected directly to the receiver or data coming from the FC but as GPS packets?
Can't we trick somehow betaflight in putting out barometer values through the GPS altitude?

https://www.team-blacksheep.com/tbs-crossfire-manual.pdf

Apologies if my wording is not the best, I'm not a native English speaker nor a developer :D

mikeller, 2021-02-25:
@wildjokerss: There simply is no field in the CRSF telemetry to return relative altitude, and there is no way to get an absolute altitude from a barometer only. Opening a feature request with TBS to have an 'altitude' telemetry frame type added might get this addressed. Or else you can buy a GPS (does not have to be TBS) and install it, or switch radio protocols.

wildjokerss, 2021-02-25:
that's pretty much what I was expecting yes, I'll see if I can log a feature request with TBS, not really sure where or how, if you have a link at hand feel free to share :D I'll look for it anyway :) Thanks for the explanation @mikeller !

wildjokerss, 2021-02-25:
okay feedback for anyone looking for this as well, I've raised a ticket with TBS, looking to see their reply

Romdrone, 2021-03-20:
Hi,
I'd love a barometer alt in tracer.!!!!
Thanks for keeping me in the loop for any info!

wildjokerss, 2021-03-20:
I the meantime I haven't had any further update from TBS, apart from their email below:

`weve added this to t he feature request  list  and will  work  on getting it added`
This was on the 2nd of March, not sure if applicable to Tracer as well.

Romdrone, 2021-03-20:
Well I guess we'll see.
👍

franz4949, 2022-04-02:
@wildjokerss and others
Isn't there any progress on TBS side for adding a barometric alt field to its CRSF protocol?
Same problem I now have with an ELRS Matek Vario receiver, which has a barometric VSpd but no barometric Alt. And Matek is pointing to the limitation of the CRSF protocol with missing Barometric altitude sensor field/ID.

VancouverUmbrella, 2022-04-02:
I'm in the same boat as @franz4949 above: The inability to get barometric altitude data from Matek vario receivers greatly reduces their usefulness for me. The same restriction applies to Matek CRSF-PWM converters.

Support at Matek informed me that, "I have seen CRSF V3 support Baro ALT, but ELRS and EDGETX don't support CRSF V3". 

Can anyone at TBS comment on whether there's a resolution in sight? 

Might any resolution depend on changes in ELRS, EdgeTx, and OpenTx?

haslinghuis, 2022-04-02:
Also see: https://github.com/betaflight/betaflight/pull/11070

haslinghuis, 2022-05-08:
Closing as inactive

VancouverUmbrella, 2022-05-08:
FYI TBS has agreed to add a baro_alt field to CRSF:

https://github.com/EdgeTX/edgetx/pull/1819
https://github.com/ExpressLRS/ExpressLRS/pull/1528\

OpenTx has also agreed to the necessary mods.

closed 2023-12-17 by haslinghuis