betaflight/betaflight #10381: Adding sim module receiver

mart1npetroff, 2020-12-03:
**Is your feature request related to a problem? Please describe.**
In our company we are developing 4G drone with SIM module. Currently we use arduino like middleware to establish communication between flight controller and sim module through MSP and AT commands.

**Describe the solution you'd like**
We could contribute and help to implement sim receiver at betaflight project.
This way we will speed-up communication between FC and SIM Module, and will get rid of arduino's weight and some extra wires.

We can help with receiver implementation and AT commands for a 4G connection and http requests.

**Describe alternatives you've considered**
Currently we use arduino for middleware and MSP.

**Additional context**
Here are some aspects that have to be discussed additionally: 
This kind of implementation needs server side and/or application to control. Currently we are developing Android App to remote control FC. We can contribute with that part too.

etracer65, 2020-12-03:
Since Betaflight has no autonomous flight capability trying to control the craft via an unreliable or variable latency cellular link would not be safe direction for development. Perhaps another firmware like INAV might be a better choice as it has position and altitude hold capability which could be triggered in the event of control signal loss or delay.

mart1npetroff, 2020-12-03:
Thank you for the idea. I will open my feature proposal at the INAV project.
Currently closing this one.

closed 2020-12-03 by mart1npetroff