betaflight/betaflight #10386: Motor PWM speed Separated from PID speed can not save successfully

ZhangBo-China, 2020-12-04:
When I chose PWM output to ESC, and enable 'Motor PWM speed Separated from PID speed' button, the default Motor PWM frequency is 480Hz, and I set 1000Hz and then press 'Save and Reboot'. When I reconnect the flight controller, the Motor PWM frequency is still 480Hz, I tried many times but all failed.

etracer65, 2020-12-04:
480hz is the maximum allowed for the PWM motor protocol. You can set it lower if you want but not higher.

mikeller, 2020-12-06:
@ZhangBo-China: The reason for the limitation to 480 Hz update frequency for PWM output is that PWM uses 1000 to 2000 us pulse length to encode the output - so at 500 Hz update frequency full output power (2000us pulses) will result in the output being high all the time, making it impossible to detect a valid signal.