betaflight/betaflight #10390: Position Hold

rgp1942, 2020-12-05:
Now that FC's have more storage, please consider putting altitude hold back into Betaflight.  I understand that racer and freestyle drones don't need no stinkin' altitude hold, but Cinewhoops do.  Restoring barometric altitude hold back into Betaflight would allow Cinewhoops to be more practical.  The purpose of this kind of drone is to take up-close-and-personal (intimate) video.  To do this requires good vertical stability when hovering and/or flying slow.  Currently all Betaflight based Cinewhoops experience "yo-yo'ing" during hover/slow speed.  If FC's had a vertical accelerometer sensor this could be controlled similar to pitch, roll, and yaw sensors.  But since FC's are 3D and not 4D, there needs to be a way to sense uncommanded vertical changes.  That's where optical sensors and barometers come in.  The state of the art in barometers is such that they can sense a 12" change.  Short of switching over to INAV, there is no way to utilize a barometer in Cinewhoop, which are all Betaflight based.  There is a need here that BF can address.  Thanks for considering!  :o)

etracer65, 2020-12-05:
The feature was removed because it never really worked. The altitude would bounce up and down and the throttle oscillated from low to high trying to correct.  Since nobody ever considered worthwhile to fix and contribute to the code (it had been this way since before even the Cleanflight days) and because it doesn't fit into the project's goals, there was no reason to keep it. So it's not a case of simply putting the feature back. The code didn't work and the rest of the project has moved on such that the original code isn't even applicable at this point. And it's a false assumption that the "FC's have more storage". F722 and F411 boards only have 512KB of space and we are frequently bumping against that.

mikeller, 2020-12-05:
This has been discussed extensively in the past:

Duplicate of #6675.

closed 2020-12-05 by mikeller

rgp1942, 2020-12-06:
OK, I didn’t realize the FC’s were still “out of memory”.  Otherwise the INAV altitude code which works great could have just been copied into BF, which would have made a lot of people like me happy.  Thanks for responding.

From: Bruce Luckcuck 
Sent: Saturday, December 5, 2020 3:28 PM
To: betaflight/betaflight 
Cc: rgp1942 ; Author 
Subject: Re: [betaflight/betaflight] Position Hold (#10390)

The feature was removed because it never really worked. The altitude would bounce up and down and the throttle oscillated from low to high trying to correct. Since nobody ever considered worthwhile to fix and contribute to the code (it had been this way since before even the Cleanflight days) and because it doesn't fit into the project's goals, there was no reason to keep it. So it's not a case of simply putting the feature back. The code didn't work and the rest of the project has moved on such that the original code isn't even applicable at this point. And it's a false assumption that the "FC's have more storage". F722 and F411 boards only have 512KB of space and we are frequently bumping against that.