betaflight/betaflight #10391: vtx IRC Tramp

enderkruemel, 2020-12-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

enderkruemel, 2020-12-09:
i have a irc trampvtx ( geprc span pro f4 v2.0 stack ) that can do 800mw, but if i use "lower power on disarm" set to on, i can only set it  to 600mw,if i arm it switches to 600mw. so far it works. if this is set to 800mw it switches no longer to the higher power but stays on 25mw. becourse i use gps and the quad stays for some time on ground to get a signal,the vtx cant set to the 800mwbecourse it overheats. and i also used pit mode but i dont like it. 25mw when the quad isnt armed is better becourse i want to see how many sattelites i get and if i can arm.

this works.
[image]

this dosnt work,it stays on 25 mw and dosnt switch to 800mw 
[image]

this works but the vtx is always on 800 mw, but whith pit mode its a workaround. (pit mode on the arm switch)
[image]

i expected that the vtx switches to from 25mw to 800mw if i arm and use low power on disarm.
 from 25 to 600 it works fine.

also the osd crosshair gets invisible becourse the artificial horizon is infront of it . that problem is on all my flightcontrollers.
[osd problem.zip](https://github.com/betaflight/betaflight/files/5665675/osd.problem.zip)
i think thats an easy fix becourse the crosshair must be in front of the artificial horizon,not behind it.

x4FF3, 2020-12-09:
irc tramp hv can do 600mW
irc tramp nano can do 500mW
pls check the corresponding manuals

enderkruemel, 2020-12-09:
thats not the problem. i know that my vtx can do 800 mw. i have the geprc span pro f4 stack (fc and vtx in one ) 
only the lower power on disarm makes problems

x4FF3, 2020-12-09:
sorry misread that. then you have to ask geprc for the correct VTX table or talk to their support about the issue.

enderkruemel, 2020-12-09:
thats a betaflight problem. the fc dosnt send the 800mw code to the vtx.

etracer65, 2020-12-09:
> thats a betaflight problem. the fc dosnt send the 800mw code to the vtx if low power on disarm is on. it sends this code only if this setting is turned off.

No, you have a setup problem and the `vtxtable` is wrong. In particular the power values are wrong. "800" is not a valid Tramp power value. So when Betaflight sends the "800" value to the vtx it rejects it and refuses to change power and stays at 25mw. Get the correct `vtxtable` settings from the manufacturer.

enderkruemel, 2020-12-09:
can you explain why the vtx can switch to 800mw  if i turn the "lower power on disarm" off? than it is correctly set to 800mw. so the vtx table is working. its only this lower power on disarm setting not the vtx table.

this sets the vtx correctly to 800. whith the lower power on disarm on it stays at 25 mw.
 https://user-images.githubusercontent.com/75631674/101626170-dd3a1a80-3a1c-11eb-98e0-7060d00e43d3.PNG

etracer65, 2020-12-09:
As I said, your `vtxtable` is wrong. Tramp protocol doesn't have a power value of 800. Try this in the CLI:
```
vtxtable powervalues 25 100 200 400 600
save
```

The "powervalues" are codes in the protocol to represent the position in the table. They are not literal power output levels. For Tramp protocol the **must be** as I've shown above. You cannot pick arbitrary values as you've done. Every vtx has different actual values for each position and you can put these in the `powerlabels`. So for your vtx 25=25, 100=100, 200=200, 400=600, 600=800.

enderkruemel, 2020-12-09:
the powermeter shows the set values are the real output ones, so also if i set it to 800 it outputs also 800 mw. so for this vtx this is the real output power. 

but also i had try this and the 400 stays at 25 mw.  i know thats wierd but i didnt know what to  do.  but the vtx works and i dont change it,it works and thats enough.

etracer65, 2020-12-09:
I gave you the solution, but ignore if you want.

enderkruemel, 2020-12-09:
I know But this is the only solution for me...

(never change a working System...)

Sorry that i ignoried it But at least i know there is something wierd ging
on whith my vtx. But i think that your settings are the right ones. So its
no Problem for the betaflight devs.

enderkruemel, 2020-12-09:
I dont know if i should let this open or if i should close it... 

But my Drohne is working for me and thats the Main Part. No big Problems i think.

closed 2020-12-10 by etracer65