betaflight/betaflight #10393: msp_displayport lost osd elements when  increased to a certain amount

ligenxxxx, 2020-12-09:
Using msp_displayport, when the number of osd elements is too large, fixed types of osd elements will be lost. Baud rate 115200.

My FC does not work as shown by etracer. When more than a dozen osd elements are turned on, it will lose some osd elements. If you choose the Artificial horizon sidebars, more than 10 osd elements may not work, and the number of missing items will be more.
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.4 Oct 20 2020 / 08:18:45 (fbcaf8c50) MSP API: 1.43
# config: manufacturer_id: DALR, board_name: DALRCF405, version: 6b1b9bea, date: 2020-05-16T15:58:54Z

# start the command batch
batch start

board_name DALRCF405
manufacturer_id DALR

# name: Divimath

# feature
feature RSSI_ADC

# map
map RTEA1234

# serial
serial 2 1 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 2 1300 1700 0 0
aux 2 2 1 1750 2100 0 0

# master
set acc_trim_pitch = 2
set acc_trim_roll = 34
set acc_calibration = 2,-53,21,1
set align_mag = CW270FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 2700
set mag_calibration = 29,-255,346
set serialrx_provider = CRSF
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set gps_sbas_mode = AUTO
set osd_tim1 = 2576
set osd_tim2 = 2593
set osd_vbat_pos = 2104
set osd_rssi_pos = 388
set osd_link_quality_pos = 256
set osd_rssi_dbm_pos = 105
set osd_tim_1_pos = 2112
set osd_tim_2_pos = 2144
set osd_remaining_time_estimate_pos = 2407
set osd_flymode_pos = 128
set osd_anti_gravity_pos = 209
set osd_g_force_pos = 2160
set osd_throttle_pos = 2347
set osd_vtx_channel_pos = 2415
set osd_crosshairs_pos = 2074
set osd_current_pos = 2130
set osd_mah_drawn_pos = 224
set osd_motor_diag_pos = 2057
set osd_craft_name_pos = 2066
set osd_display_name_pos = 2048
set osd_gps_speed_pos = 131
set osd_gps_lon_pos = 333
set osd_gps_lat_pos = 49
set osd_gps_sats_pos = 32
set osd_home_dir_pos = 43
set osd_home_dist_pos = 80
set osd_flight_dist_pos = 179
set osd_compass_bar_pos = 2088
set osd_pid_roll_pos = 2208
set osd_pid_pitch_pos = 2240
set osd_pid_yaw_pos = 2272
set osd_debug_pos = 2176
set osd_power_pos = 2289
set osd_warnings_pos = 2438
set osd_avg_cell_voltage_pos = 2098
set osd_pit_ang_pos = 2122
set osd_rol_ang_pos = 2154
set osd_battery_usage_pos = 224
set osd_disarmed_pos = 2080
set osd_nheading_pos = 2226
set osd_nvario_pos = 2263
set osd_esc_tmp_pos = 175
set osd_esc_rpm_pos = 387
set osd_esc_rpm_freq_pos = 391
set osd_rtc_date_time_pos = 136
set osd_adjustment_range_pos = 269
set osd_flip_arrow_pos = 212
set osd_core_temp_pos = 2168
set osd_stick_overlay_left_pos = 2323
set osd_stick_overlay_right_pos = 2272
set osd_efficiency_pos = 181
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_displayport_device = MSP
set vcd_video_system = NTSC
set displayport_msp_serial = 2
set name = Divimath

profile 0

rateprofile 0

# end the command batch
batch end
```

Flight controller [DALRCF405]

Can someone use the above settings to try it on their FC, grab the hex data through the serial port tool and save it as a txt file and share it with me?
I want to determine if I am the only one who has this error, and if there are other settings I want to change.
I set up as follows:
Port:
    msp enabled, speed115200
CLI:
    set osd_displayport_device = MSP
    set displayport_msp_serial = 2
    save

haslinghuis, 2020-12-09:
Please provide the URL for the serial port tool as mentioned

ligenxxxx, 2020-12-09:
https://cdn.eltima.com/download/freeserialmonitor.exe
A simple tools.

haslinghuis, 2020-12-09:
There is no output at all?

Side note: Also found out that will latest firmware development build the Font Manager does not work and the image in configurator does show only the top.

Anyway tried with latest nightly configurator and latest release and latest development build. Also tried using different speed and flow control.

ligenxxxx, 2020-12-10:
CLI:
set osd_displayport_device = MSP
set displayport_msp_serial = 'x'
save
The value of'x' determines the serial number of the msp you are using. If you are using serial port 3, then x=3-1=2.
Another point is that USB_VCP output cannot be used.

haslinghuis, 2020-12-10:
> Another point is that USB_VCP output cannot be used.

Using USB_VCP here and don't have a FrSky OSD. But the DALRCF405 also has USB_VCP. Are you using FTDI adapter or am I missing something?

ligenxxxx, 2020-12-11:
Another point is that USB_VCP output cannot be used  as a displayport_msp_serial. sorry,I did not describe clearly.
In addition to flight controllers and computers, additional hardware is required to collect data.
You need a serialPort_to_usb device.
I made a simple software that can display the msp_displayport received by the serial port in real time.
If you have a serialPort_to_usb device, I can send the software to you for a more intuitive view of the output of osd elements.

haslinghuis, 2020-12-11:
Yeah have a FTDI adapter here. But the USB connector of my spare testing FC is broken, and not going to rip my working quad's apart. Have not decided yet on my next build. But probable can get a test FC if asking my friends soon.

ligenxxxx, 2020-12-12:
Man, I will remember you. F7 is great. I haven't tested it on f7. If you succeed, maybe you can explain some problems.

mikeller, 2020-12-12:
@ligenxxxx:

> `serial 2 1 115200 57600 0 115200`

This is too slow for MSP displayport when used for in-flight display - increase the UART speed.

haslinghuis, 2020-12-12:
Hooked up FTDI adapter to UART1 => not able to flash¿. Starting in DFU using boot button did it or maybe I just was lucky that the clip on the USB port is exactly in place but I think it has lost connection while flashing the last time. Anyway have now two connections tried to raise the MSP on UART1 to 1000000 but serial Port Monitor does still not log anything. Could you please share your `tool` to look at it.

ligenxxxx, 2020-12-14:
> Hooked up FTDI adapter to UART1 => not able to flash¿. Starting in DFU using boot button did it or maybe I just was lucky that the clip on the USB port is exactly in place but I think it has lost [...]

I send it to your email mark@numloq.nl

haslinghuis, 2020-12-14:
[image]

ligenxxxx, 2020-12-14:
Congratulations,！can you try to add more elements?  Or just use the settings I provided before.  As far as I know, more than 14 may be easily lost

haslinghuis, 2020-12-14:
This is the output with the osd settings taken from your diff. Adding more elements does not change the output

mikeller, 2020-12-14:
@haslinghuis: Try

```
serial 2 1 500000 57600 0 115200
```

and see if this is working when an appropriate port speed is used.

haslinghuis, 2020-12-14:
Already tried on 1000000 baud.

ligenxxxx, 2020-12-14:
Conclusion: On f7, 100000bps, still unable to transmit a large number of osd elements.

ligenxxxx, 2020-12-15:
I did some debugging.
I printed the sum of mspSerialEncode() return value ret every time between clearScreen() and drawScreen(), and found the following rules:
When the serial port baud rate is 500000bps, sum_max = 280; 
when 250,000bps, sum_max=271; 
when 115200bps, sum_max=263; 
when 19200bps, sum_max=160, 
when 9600bps, sum_max=80.
Why is the amount of data transmitted by the serial port not proportional to the speed of the serial port?
I will do further testing next.

ligenxxxx, 2020-12-15:
when osd elements is lost, mspSerialEncode return ret < datalen.

ligenxxxx, 2020-12-15:
After I added a line of code, the problem was solved, but I don't know if it makes sense.
msp/msp_serial.c line579 add:
while(ret<datalen)

closed 2021-02-15 by github-actions[bot]