betaflight/betaflight #10394: Pilot Heading Arrow

buzkill420, 2020-12-09:
Was thinking how nice it is having direction to home arrow on the osd, an necessary for any long range flight. Flying my DJI, i also find it very handy to have the direction I'm facing onscreen as well, allowing you to position yourself for optimal reception..ie goggles and transmitter.
Was thinking how betaflight might be able to accomplish the same thing. 

Using a heastracker (I'm using the internal fatshark module)  you should be able to use the gyroscope to track your position by feeding the correct axis through a channel to your fc. If only using the internal module, it lacks a compass, so auto aligning is out (and tbh idk if the compass data is even output, far over my head) so a "heading set" mode would have to be created- just like headless mode and headajust.

What do you think?  Worth wild or just a waste of time? Im curious what others think of the idea.

wxyzsupermod, 2021-01-06:
Thats an interesting Idea for sure, but for the small amount of head tracking users its probably not worth. I would say it might be better for you to construct a base station, with a servo controlled antenna tracker if that's what you want.

mikeller, 2021-01-10:
Putting yourself into the situation where you have to multitask by controlling your craft's flight path **and** controllong your head position simultaneously based on on-screen information sounds like a recipe for crashes and accidents.
I think @wxyzsupermod's suggestion to get yourself a base station with an antenna tracker for long range flight is much better - they can normally be built from things like old servos and flight controllers that you can find in your spare parts box, or in the discount bin of your local RC store.

buzkill420, 2021-01-10:
> Putting yourself into the situation where you have to multitask by controlling your craft's flight path **and** controllong your head position simultaneously based on on-screen information sounds [...]

Though i agree with both of you on the base station idea,  i don't agree with you on it being a "receipe  for a crash." Quite the contrary,  I'm currently moving my head with NO reference frame- apart from memory and guessing- to find the best signal and have no issues with crashing ;p  if no one thinks this is a fun idea, I'll be the first to say let's close this.  I just thought it would be a fun thing to try, nvm very useful if managed to get working.

closed 2021-02-16 by github-actions[bot]