betaflight/betaflight #10395: osd problem crosshair & artificial horizon

enderkruemel, 2020-12-09:
also the osd crosshair gets invisible becourse the artificial horizon is infront of it . that problem is on all my flightcontrollers.
i think thats an easy fix becourse the crosshair must be in front of the artificial horizon,not behind it. 

here is a dvr recording whith the lens covered.(to see the osd better) also i use only a 2s for testing not a 6s. 
[osd problem.zip](https://github.com/betaflight/betaflight/files/5665760/osd.problem.zip)

i use a mobula7 (F4 fc) and 
a geprc crocodile7 longrange (runcam split, span pro f4 v2 stack)

the problem startet as i upgradet my fc to the newest version and i get only v on the screen. so i uploadet the font in the osd tab.  
( i use betaflight 4.2.5)

i know that the problem is only software and its in the osd. i think a diff or a dump file isnt helpfull at all.
[get osd.txt](https://github.com/betaflight/betaflight/files/5665805/get.osd.txt)

McGiverGim, 2020-12-17:
This was fixed here: https://github.com/betaflight/betaflight/pull/9949 but it has been posponed until version 4.3.

You can use a development version of 4.3 if you are comfortable with it.

enderkruemel, 2020-12-17:
Ok thanks,  i didnt See it

closed 2020-12-17 by enderkruemel