betaflight/betaflight #10396: Custom curves on rate profiles

lacasitoFPV, 2020-12-09:
Good morning,

I've been thinking a lot on this and I would like to know if someone thought already about this or not. Lately I played a lot with the rates profiles trying to find something that fits to my flight style, but I couldn't find any combination that matches what I want. Instead, as I did on my helicopters, I decided to introduce the curves of the transmitters into the rates equation. The results are pretty nice and somehow they cover my needs.

The question is, have you ever thought on letting the pilot to make his own curve instead of controlling the curve with the RC, Super and Expo rate parameters? At the end, you need a specific curve depending on the flight performance, and with the equations of Betaflight is not possible to make all kind of curves. And I think that is why on throttle curves or the rates for the helicopters, you always use the curves on the transmitter.

The only thing you need is maximum deg/sec on each axis and then make the curve you want with different points.

So that, my proposal is to add this option into Betaflight. What do you think? Who I have to talk to so that we can do some trials?

Kind regards!

LacasitoFPV

[image]

klutvott123, 2020-12-09:
Have you tried using ```ACTUAL``` rates? https://github.com/betaflight/betaflight/pull/9495

lacasitoFPV, 2020-12-09:
> Have you tried using `ACTUAL` rates? #9495

Yes, I tryed but it's not possible to make linear the curve at the end and exponential at the beggining. That's why making the curve with points like on the transmitter, could be great!

Kind regards,

Lacasito

mikeller, 2020-12-11:
@lacasitoFPV: Yes, this has been discussed in the past, and parametric curves for rates have been found to be desirable in Betaflight.
Just open a pull request for your proposed implementation, and I am sure there will be volunteers to trial it.

lacasitoFPV, 2020-12-11:
> @lacasitoFPV: Yes, this has been discussed in the past, and parametric curves for rates have been found to be desirable in Betaflight.
> Just open a pull request for your proposed implementation, and I am sure there will be volunteers to trial it.

How can I open a pull request? That would be wonderful! Thanks a lot!

mikeller, 2020-12-11:
@lacasitoFPV: Here is the basic workflow: https://gist.github.com/Chaser324/ce0505fbed06b947d962 - assuming that you already have a working implementation of parametric curves for Betaflight that you would like to trial.

lacasitoFPV, 2020-12-11:
Right know what I'm doing is 

> @lacasitoFPV: Here is the basic workflow: https://gist.github.com/Chaser324/ce0505fbed06b947d962 - assuming that you already have a working implementation of parametric curves for Betaflight that you would like to trial.

Right know what I'm doing is making a total linear rate profile in Betaflight and then on the transmitter I put the curve I want with as many points as I need. I also have a simple excel file where I can check my points with the curve projected.

mikeller, 2020-12-11:
@lacasitoFPV: Fair enough - you can always open a pull request once you have done the implementation in the firmware in order to get this trialled.

ctzsnooze, 2020-12-29:
Actual rates is essentially two straight lines with a curved section in-between.
It is possible to get a nearly straight line at higher rpm, with most of the curvature at the bottom:
[image]
The centre can be made quite straight, with most of the curvature at the top:
[image]

closed 2021-02-15 by github-actions[bot]