betaflight/betaflight #10397: ERROR: diff: NO CONFIG FOUND - STM32F405

ONESucH, 2020-12-09:
# diff all
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F405 (S405) 4.2.5 Nov 22 2020 / 18:38:23 (afdac08b3) MSP API: 1.43
###ERROR: diff: NO CONFIG FOUND###
# start the command batch
batch start

# reset configuration to default settings
defaults nosave

mcu_id 002e003e3339471730353636
signature 

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

Пробовал бегать по разным версиям ПО, прошивкам так ни к чему и не пришел.
Можете подсказать что нужно прописать в CLI, чтобы довести прошивку до рабочего вида?

На плате STM32F405 горит красная лампочка.

haslinghuis, 2020-12-09:
You are supposed to choose a unified target instead of a MCU target. Please flash the target using the target specified by the manufacturer of your FC

ONESucH, 2020-12-09:
> Вы должны выбрать единую цель вместо цели MCU. Пожалуйста, прошейте цель, указанную производителем вашего FC

Ничего не понял ;0
"единую цель" какую цель?
Что такое "MCU"
"прошейте цель, указанную производителем вашего FC" - то есть через CLI или готовую прошивку?

haslinghuis, 2020-12-09:
You need to know the brand, type and version of the flight controller. You can ask your supplier which firmware you need for your flight controller. 
Also this is not the place for users to ask questions about configuration, but for solving software problems or suggesting new ideas. Please try other resources.

ONESucH, 2020-12-10:
> https://github.com/betaflight/betaflight/wiki/Installing-Betaflight

Извиняюсь что нахожусь не в нужной теме, но у меня выбора нету, информации мало, помочь не кому что делать не знаю.
Убил много времени так и не решил проблему.
Попробовал пройти по вашей ссылке "https://github.com/betaflight/betaflight/wiki/Installing-Betaflight", выполнил всю инструкцию, после прошивки результата никакого.
Попробовал еще раз повторить - ничего.

Совсем не знаю что делать.

ONESucH, 2020-12-10:
Ураа, я решил проблему)

ONESucH, 2020-12-10:
Спасибо всем за терпение ;)

ONESucH, 2020-12-10:
Если нужны ответы мои, отпишу без проблем ;)

closed 2020-12-10 by ONESucH

haslinghuis, 2020-12-10:
Я рад, что вы смогли решить свою проблему!

ArtemUnreal, 2024-06-25:
> Если нужны ответы мои, отпишу без проблем ;)

Можете описать, как вы решили проблему