betaflight/betaflight #10400: Betaflight 4.2.X - Barometer reads constantly decreasing values

moserello, 2020-12-10:
**Describe the bug**
I have a Kakute F7 v1.5 with integrated BMP280. Altitude readings continue to decrease after the quad is plugged in. 
I'm aware of #5311 but in this case readings can reach -800m on the bench.
After the insertion of the LiPo the altitude descends fast. After some minutes, if I unplug the LiPo leaving just the USB, readings start to increase a little

[image]

plugged in the LiPo
[image]

I've tried to flash all the 4.2.X available versions for both the KAKUTEF7 and KAKUTEF7V2 targets but nothing seemed to change.

```

# 

# dump

# version
# Betaflight / STM32F745 (S745) 4.2.5 Nov 22 2020 / 18:39:28 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7V2, version: ee5e8acf, date: 2020-02-02T04:22:29Z

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# name: RAZORWHIP

# resources
resource BEEPER 1 D15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 C09
resource MOTOR 6 A03
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 E13
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
[... 1045 lines cut ...]
set yaw_rc_rate = 130
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 

```

Is there a way to prevent it, maybe calibrating the sensor or something similar?
Maybe a full chip erase update deleted the factory baro calibration?

etracer65, 2020-12-10:
First of all, the instructions request that you provide a `diff` of your settings. The `dump` output is not useful. Secondly it sounds like you have a failing barometer or possibly failing voltage regulator circuits that supply the onboard components. No there is nothing that Betaflight can do to correct this.

moserello, 2020-12-10:
> First of all, the instructions request that you provide a `diff` of your settings. The `dump` output is not useful. Secondly it sounds like you have a failing barometer or possibly failing voltage [...]

I'm sorry for the dump instead of the diff, my bad at all. 
The fact is that this sensor starts to invert the trend of readings when lipo is disconnected (maybe for temperature reasons?), so I hoped that I could calibrate it. Obviously I can't if the barometer of BEC are broken. 
Is there a way to test where the problem is, in order to exclude software issues?

etracer65, 2020-12-10:
There are no settings or mechanism in Betaflight to "calibrate" the behavior of the barometer sensor. As far as troubleshooting you would probably need an oscilloscope and schematic for your flight controller along with understanding the datasheet for the BMP280. But ultimately I'm not sure what that would accomplish. Even if you could isolate it to a power vs. sensor problem it's unlikely you could repair the board anyway. Also since the BMP280 is the most popular sensor and is used on many boards with many many users having no problems, it seems likely we can rule out a software issue. But since you haven't provided the requested settings `diff` there's no way for us to verify you haven't changed some setting that's causing the problem.

moserello, 2020-12-11:
> There are no settings or mechanism in Betaflight to "calibrate" the behavior of the barometer sensor. As far as troubleshooting you would probably need an oscilloscope and schematic for your flight [...]

Thanks for the reply, and I ask sorry for the missing `diff`: I've just updated with the `diff` instead of the `dump`. I agree that, since it seems to be a hardware problem, troubleshooting doesn't worth enough. 
About BMP280 I've others on other boards that are working very well. 

Thanks for the patience, I've updated diff. If nothing is set wrongly on SW side, I will close this issue with the next comment.

etracer65, 2020-12-11:
> Thanks for the patience, I've updated diff. If nothing is set wrongly on SW side, I will close this issue with the next comment.

I'm not sure how this is the correct settings `diff`. It shows that you have the barometer completely disabled so there's no possible way it can work at all.
```
set baro_hardware = NONE
```
Also why do you have this setting change?
```
set baro_noise_lpf = 900
```
The default is 600 and you shouldn't change this. It's doubtful that this would cause the problems you're having but it's not a setting you should change.

Also, this is a corrupted setting:
```
set gyro_2_spibus = 4
```
I'm not sure why you would ever change this. That board doesn't have a second gyro and this setting should be 0.

Finally, it seems like your config is just generally messed up. You seem to have a unified config flashed for the `KAKUTEF7V2` but the settings are configured for the target `KAKUTEF7`.
```
# config: manufacturer_id: HBRO, board_name: KAKUTEF7V2, version: ee5e8acf, date: 2020-02-02T04:22:29Z
board_name KAKUTEF7
```
Since you seem to have some settings changes that make no sense it would probably be a good idea to flash the proper target (contact the manufacturer) and be sure to enable "Full chip erase". Then set up from scratch (do not paste the `diff` back in).

moserello, 2020-12-16:
> I'm not sure how this is the correct settings `diff`. It shows that you have the barometer completely disabled so there's no possible way it can work at all.
>
> ```
> set baro_hardware = NONE
> ```
>
> Also [...]

Thank for the reply.
Actually you are right. I uploaded the "last configuration" taking into account that the barometer was broken. 

This is why I disabled it. 
The low pass filter was increased from default value because it seemed to reduce variance. Unfortunately there was a decreasing trend that was impossible to "filter out". 

`set gyro_2_spibus = 4` this is curious but default is 4 (seen on cli). I think it is related to the target. Don't know why it could be 4 instead of 0. But my question here is: if the spibus was set uncorrect, i wouldn't have seen any baro connected to the FC. 

For what concerns the the target, I have a kakuteF7 V1.5 and it is not yet available any kakuteF7 V2 board. I read that the v2 correspond to the upgraded version of the kakute V1, with some differences on pins assignations (here could be explained the previous point about `gyro_2_spibus`). On the manual you can find both targets in different points of the documentation, without any explanation about differences.

Both targets seem to show the same problem on barometer, but doesn't show any other difference on my FC. 
Here it is my last diff (problem still persists):

```# diff

# version
# Betaflight / STM32F745 (S745) 4.2.5 Nov 22 2020 / 18:39:28 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# name: RAZORWHIP

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 3 32 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 4 1800 2100 0 0
aux 2 2 4 1300 1700 0 0
aux 3 27 1 1800 2100 0 0
aux 4 46 8 1300 2100 0 0
aux 5 13 2 1600 2100 0 0
aux 6 13 3 1600 2100 0 0
aux 7 35 9 1600 2100 0 0
aux 8 39 5 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 800

# master
set acc_calibration = -79,-56,-120,1
set max_check = 1950
set rssi_channel = 16
set fpv_mix_degrees = 20
set serialrx_provider = SBUS
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 35
set gps_rescue_initial_alt = 100
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1200
set gps_rescue_throttle_max = 1700
set gps_rescue_throttle_hover = 1250
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_warn_over_cap = ON
set osd_rssi_pos = 2554
set osd_tim_2_pos = 2081
set osd_throttle_pos = 2505
set osd_ah_pos = 2126
set osd_mah_drawn_pos = 2097
set osd_gps_speed_pos = 2511
set osd_gps_lon_pos = 2540
set osd_gps_lat_pos = 2528
set osd_gps_sats_pos = 2137
set osd_home_dir_pos = 2523
set osd_home_dist_pos = 2518
set osd_flight_dist_pos = 470
set osd_altitude_pos = 2089
set osd_avg_cell_voltage_pos = 2104
set osd_disarmed_pos = 2313
set osd_esc_tmp_pos = 456
set osd_core_temp_pos = 2498
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_esc_temp = ON
set osd_stat_flight_dist = ON
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5732
set gyro_1_align_yaw = 2700
set name = RAZORWHIP

profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 130
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 75
set pitch_srate = 75

# end the command batch
batch end
```

I think I will buy an external barometer in order to see if it fix the issue

closed 2021-02-15 by github-actions[bot]