betaflight/betaflight #10404: PIDs tuned for 150Hz RC link rate are inadequate for 50hz rate (CSRF)

ladislawv, 2020-12-14:
Setpoints are smoothed well in 50hz mode, however there is a slight ripple which ends up amplified on the motors.

Is there a solution to this problem other than just tuning PIDs for a lower rate?

Having an option to tune PIDs for each RC link rate might be the solution. Perhaps just dynamically scaling P & D gain could work well.

[50hz 150hz problem.zip](https://github.com/betaflight/betaflight/files/5690856/50hz.150hz.problem.zip)

[image]

closed 2020-12-14 by ladislawv

etracer65, 2020-12-14:
LOL! You realize that we can still see the original subject and content right?

ladislawv, 2020-12-15:
Content is deleted. Subject is not clearly visible anymore :D
RC smoothing was fixed to 20ms by mistake. Traces look equally bad at both rates with default RC smoothing.