betaflight/betaflight #10411: Barometer not recognized after update to firmware version 4.2.5

PieterK95, 2020-12-16:
Bug description:
I recently bought an MATEK F722-SE Flight controller. When I plugged it into the computer, betaflight recognized the barometer. Then i decided to update the firmware from 4.2.2 to 4.2.5. After this firmware update, betaflight did not recognized the barometer anymore. The flightcontroller is not wired up yet.

What can cause this issue? 

Thanks in advance!

DIFF ALL:

diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.5 Nov 22 2020 / 18:39:53 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722SE, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722SE
manufacturer_id MTKS
mcu_id 00170028465350102030364e
signature 

# master
set acc_calibration = 109,-61,228,1
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
Write your command here. Press Tab for AutoComplete.

etracer65, 2020-12-16:
I just tested 4.2.5 on a Matek F722-SE and the barometer works normally. Try going back to an older version to see if you still have a problem. Maybe your barometer has failed.

PieterK95, 2020-12-17:
Hi etracer 65,

Thank you for the quick response.  I already tried a couple of older firmware versions, unfortunately the barometer is also not recognized in these versions. Do I need battery voltage to make the baro work properly? The LIPO is not connected to the flightcontroller yet.

This flightcontroller is brand new and not wired up yet. Does barometer DOA happens more often?

Thanks for the help!

Regards,
Knoops Pieter

PieterK95, 2020-12-17:
Hi,

The issue is solved. I have the MATEK F722-SE with DPS310 baro. The firmware from the betaflight configurator does not support this baro. Because of this reason i have to download the correct firmware from the MATEK website.

[image]

link to the site: http://www.mateksys.com/?portfolio=f722-se#tab-id-8

closed 2020-12-17 by etracer65

ghostkcleung, 2021-02-09:
So, would the official betaflight implement the DPS310 in the future ?
I only believe the official firmware.

etracer65, 2021-02-09:
> So, would the official betaflight implement the DPS310 in the future ?
> I only believe the official firmware.

Yes, in 4.3.