betaflight/betaflight #10412: [Feature Request] Add "Disarm On Crash" functionality

TyLauriente, 2020-12-16:
#### **Is your feature request related to a problem? Please describe.**
The **problem** is that when a **drone crashes**, it will continue to **spin the motors until it is disarmed**.

#### **Describe the solution you'd like**
I would like an option in Betaflight to **disarm the quad when it crashes**. 
I'm not sure exactly how this would look on the implementation side. (I'm new to Betaflight).
Maybe a potential solution could be to compare the **Motor RPM** to an **expected value** and if it dips bellow a **thresh-hold**, **disarm the quad**.

#### **Describe alternatives you've considered**
N/A

#### **Additional context**
I'm sure a lot of people (especially beginners) would appreciate a feature like this.

mikeller, 2020-12-16:
@TyLauriente: You're lucky. This sounds exactly like the 'crash recovery' feature: https://intofpv.com/t-crash-recovery-mode-useful-for-whoops

TyLauriente, 2020-12-16:
That's fantastic, thank you!

closed 2020-12-16 by TyLauriente