betaflight/betaflight #10414: Mamba FC F405 DJI don't support all serial provider protocol

padel-fr, 2020-12-18:
### Describe the bug
I'm recently the owner of new Diatone Roma F5 (FC F405DJI) and I would like to link the FC with my Jeti receiver.
I need to use a jeti receiver plugged in the TX1 pad to control the quad and to receive the telemetry information. I have the same setup with HGLRC FC F722 and it's working very well. it's not the same sound side F405DJI FC then do you know if i have to apply a specific setup under betaflight to work fine?

I contacted Diatone support but they are no knowledge on theirs products.

### To Reproduce

The issue is easily reproducible in Jeti Serial provider mode. If i switch my receiver in Sbus mode, plug on RC pad it's work fine but without telemetry support. There is i also two other TX1 Pad (inverted and non inverted) but it stay not working anymore.   

### Expected behavior

I don't understand how Mamba can limit the support of serial provider on this FC !!! I would like just use the benefits of Jeti protocol. 

### Flight controller configuration
# 

# dump all

# version
# Betaflight / STM32F405 (S405) 4.2.5 Nov 22 2020 / 18:38:23 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: MAMBAF405US, version: 45b6e90d, date: 2020-11-27T19:02:33Z

# start the command batch
batch start

board_name MAMBAF405US
manufacturer_id DIAT
mcu_id 0041001f3053500120303948
signature 

# name: DIATONE ROMA

# resources
resource BEEPER 1 C13
resource MOTOR 1 A09
resource MOTOR 2 A08
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B09
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 B03
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
[... 1374 lines cut ...]
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

### Setup / Versions

Betaflight / STM32F405 (S405) 4.2.5 Nov 22 2020 / 18:38:23 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: MAMBAF405US, version: 45b6e90d, date: 2020-11-27T19:02:33Z

 - How are the different components wired up:

Receiver connected on RC pad, tried to connect also on TX1 pad

### Additional context

please find herewith the diagram of FC.
[image]

Guili123, 2020-12-18:
I am not sure if this is related to the same issue/bug. But if yes:

I have the same Roma F5 /Mamba 405 DJI FC, and don`t get telemetry with Ghost on BF 4.3 2356.
The same software setup works with the HGLRC F722 setup without issues.
The RX is working with SRXL2 and GHOST on the Mamba 405DJI, but on Ghost no Voltage/Current or mah used is send back to the TX. (Ghost protocol is needed for telemetry)

padel-fr, 2020-12-18:
Thank you for your feedback, but in my case i have not remote control not telemetry....remote control works only with serial provider Sbus mode setup but i don't understand why Mamba products would limit the protocol support on theirs FC when Betaflight is fully compatible with many serial providers. if anybody has an idea ?

McGiverGim, 2020-12-18:
Please edit your initial message and replace the `dump all` by a `diff` and a `resource show all` as suggested in the template that is shown when opening the issue. The `dump all` gives too much information to be reviewed.

etracer65, 2020-12-18:
The F4 MCU can't communicate with inverted serial, while a F7 MCU can. If you want to use it with F4 then you'll need to get a non-inverted source from your receiver if possible or use a flight controller that has a bidirectional inverter (typically for S.port support). The SBUS pad is a unidirectional inverter and won't work. Or just use a F7 flight controller that can handle the inverted signal.

padel-fr, 2020-12-18:
I have another F4 MCU Mamba F405 with Betaflight 3.5.7 which works very fine, but i know you have change the Betaflight code for Jeti protocol to use directly a TX pad from Betaflight 4.x. Do you know if it could be the root cause ?
 
[How.to.setup.your.RX.and.FC.for.Jeti.Ex-Bus.pdf](https://github.com/betaflight/betaflight/files/5715911/How.to.setup.your.RX.and.FC.for.Jeti.Ex-Bus.pdf)

mikeller, 2020-12-18:
@padel-fr: What UART is the Jeti RX connected to on the other Mamba F405? A `diff` for this one will help as well.

Guili123, 2020-12-18:
@padel-fr
I found my issue. Try to switch the RX to any other TX pad.
It seems the inverted and univerted TX1 from this Mamba F405US can NOT send telemetry.
I was receiving telemetry after I switched to TX3!
May be this helps?

padel-fr, 2020-12-19:
Hello, thanks for your feedback @Guili123. This morning i tried all free TX UART (TX2,TX4,TX5) on the F405US and the setting with serial rx provider JETI works for the commands but nothing regarding the telemetry. It seems TX1 is a specific pad !!!! If any developer can be check why the telemetry output don't work, i will be cool...

Guili123, 2020-12-19:
@padel-fr Have you tried both TX1 pads? Because there is one inverted TX1 and one uninverted TX1 pad.
TX1 is definetely special, because I couldn`t receive telemetry where telemetry works on TX3 for me. Haven`t tested the TX pads you mentioned.

padel-fr, 2020-12-19:
@Guili123 Yes, i tried both TX1 pads but in this case no command and no telemetry. In any case Jeti signal is univerted signal.

mikeller, 2020-12-19:
@Guili123:

> It seems the inverted and univerted TX1 from this Mamba F405US can NOT send telemetry.

Depending on the design of the external inverter this is possible. Since Diatone does not publish the schematics for their hardware there is no way to tell if this should work or not.

> If any developer can be check why the telemetry output don't work, i will be cool...

Have you ever considered contacting the manufacturer or your supplier for help with your hardware problem? After all, it is them who got the money you paid for the hardware.

padel-fr, 2020-12-20:
Hello everybody, I found a workaround. I just switched to Emuflight and now everything works perfectly.

closed 2020-12-20 by etracer65

ramgolams, 2021-02-15:
I'm having this same issue. I'm using an RXS-R and I cannot get F.PORT to work. I tried the inverted and non-inverted TX1, and even TX3. I tested my receiver on another drone and it works perfectly fine. The only way I can get it to work on this board is to set it to SBUS and reflash the receiver to SBUS mode. There is something up with this board or the BF build.

mikeller, 2021-02-16:
@ramgolams: For FPort you need a *bidirectional* inverted port - not sure this board has one, check the manufacturer's manual. The alternative is to do an 'uninversion' hack to your TX to get the uninverted FPort signal, and feed this into a normal TX pad on the flight controller.

ramgolams, 2021-02-16:
> @ramgolams: For FPort you need a _bidirectional_ inverted port - not sure this board has one, check the manufacturer's manual. The alternative is to do an 'uninversion' hack to your TX to get the uninverted FPort signal, and feed this into a normal TX pad on the flight controller.

Thanks for the tip. That's the only thing I didn't try as I lifted the tiny pad on the receiver in the processes. Will just stick to SBUS for now.