betaflight/betaflight #10420: MSC on MATEKH743 not working

TheIsotopes, 2020-12-22:
### Describe the bug
after activating msc mode, msc is initialized and then stops abruptly
no new drive is displayed on the pc, fc appears to be frozen
i tested this with three different sd cards

@SteveCEvans can you check what's wrong

### To Reproduce
using all builds since supporting MATEKH743

### Flight controller configuration

```
# diff

# version
# Betaflight / MATEKH743 (M743) 4.3.0 Dec 22 2020 / 10:01:26 (3f116fd10) MSP API: 1.44

# start the command batch
batch start

board_name MATEKH743

profile 0

rateprofile 0

# end the command batch
batch end

# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: FREE
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: BEEPER
B00: FREE
B01: FREE
B02: GYRO_EXTI
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: FREE
C03: FREE
C04: ADC_EXT
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: SDIO_D0
C09: SDIO_D1
C10: SDIO_D2
C11: SDIO_D3
C12: SDIO_CK
C13: FREE
C14: FREE
C15: GYRO_CS 1
D00: FREE
D01: FREE
D02: SDIO_CMD
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: SPI_MOSI 1
D08: FREE
D09: FREE
D10: PINIO 1
D11: PINIO 2
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: USB_DETECT
E03: LED 1
E04: LED 2
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: GYRO_CS 2
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: BEEPER
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

SteveCEvans, 2020-12-22:
@TheIsotopes Just tested current master (`3f116fd10`) on an SPRACINGH7NANO, the only H7 based FC I have with SD card and that's working fine. The config in target.h looks very similar except that SDCARD_DETECT_INVERTED is defined on the SPRACINGH7NANO. I'd be surprised if there's an inverter fitted anywhere, so maybe the sockets used by each have different default open/closed behaviours? Maybe a long shot, but could you please try adding `SDCARD_DETECT_INVERTED` to the MATEKH743 target.h?

Does logging work OK? I presume the card is detected OK.

### SPRACINGH7NANO
```
#define USE_SDCARD
#define USE_SDCARD_SDIO
#define SDCARD_DETECT_PIN PD10
#define SDCARD_DETECT_INVERTED
#define SDIO_DEVICE             SDIODEV_1
#define SDIO_USE_4BIT           true
#define SDIO_CK_PIN             PC12
#define SDIO_CMD_PIN            PD2
#define SDIO_D0_PIN             PC8
#define SDIO_D1_PIN             PC9
#define SDIO_D2_PIN             PC10
#define SDIO_D3_PIN             PC11
```

### MATEKH743
```
#define USE_SDCARD
#define USE_SDCARD_SDIO
#define SDCARD_DETECT_PIN       NONE
#define SDIO_DEVICE             SDIODEV_1
#define SDIO_USE_4BIT           true
#define SDIO_CK_PIN             PC12
#define SDIO_CMD_PIN            PD2
#define SDIO_D0_PIN             PC8
#define SDIO_D1_PIN             PC9
#define SDIO_D2_PIN             PC10
#define SDIO_D3_PIN             PC11
```

TheIsotopes, 2020-12-22:
> Does logging work OK? I presume the card is detected OK.

SD card is detected and logging also works.

> ... but could you please try adding SDCARD_DETECT_INVERTED to the MATEKH743 target.h?

it does not work, fc does not boot anymore with inserted sd card.

MATEKSYS, 2020-12-23:
I tested 2G and 32G SD card on H743-SLIM with latest 4.3.0#2364 just now.
MSC works normal

MATEKSYS, 2020-12-23:
[image]
[image]

I tested 2G and 32G SD card on H743-SLIM with latest 4.3.0#2364 just now.
MSC works normal

TheIsotopes, 2020-12-23:
@MATEKSYS Should that really be due to the SD cards?
Then the fc is pretty picky. What brand of cards are you using?

MATEKSYS, 2020-12-23:
[image]

TheIsotopes, 2020-12-23:
ok, then it seems to be due to my sd cards.

thanks again 👍

closed 2020-12-23 by TheIsotopes

mikeller, 2020-12-26:
@TheIsotopes: I think I have to weigh in on @MATEKSYS' side here - a lot of (if not all) flight controllers seem to be picky about SD cards. Not sure what it is that seems to be making us more susceptible to problems than other devices, but the reality is that we are.

pichim, 2022-01-07:
@TheIsotopes @MATEKSYS I had a similar issue. I'm running a MatekSysH743-Slim and a MatekSysH743-Mini with BF4.3RC1 and i'm using 2 different kind of SD cards ScanDisk Ultra HCI and Transcend Ultimate 800x 16 gb HCI. The ScanDisk is working on both FC (Mini and Slim) in 8k pidloop / 2k logging where the Transcend is only working properly on the MatekSysH743-Mini. So it really was related to the SD card type. Just in case any one has the same problems like I did.

pichim, 2022-01-30:
@TheIsotopes  @MATEKSYS hi guys, here an update. i am actually also facing logging issues with the matek h743 family. im using a recent nightly build: betaflight_4.3.0_MATEKH743_b6bc232a1.hex
i have 3 boards an the logging is working properly only on the mini, see below:
matek h743 mini: both sd card ok (Transcend Ultimate 800x 16 gb HCI, ScanDisk Ultra 16 gb HCI)
matek h743 slim: scan disk can log but skippes datapoints, transcend not working at all
lumenier lux h7 hd (they refer to matek slim): both cards not working

pichim, 2022-02-20:
@MATEKSYS Using  the ScanDisk Ultra 16 gb HCI and adding a Mamba Interference Isolation Board V2 solved the issue for the matek h743 slim