betaflight/betaflight #10423: Soft serial RPMFILTER problems....

berendb0tje, 2020-12-24:
### Describe the bug
When i use the softserial tab in betaflight configurator. I get the RPMFILTER and D-shot BBANG message. So when i turn on soft serial i can't use the smart audio. 

### To Reproduce
Turn on softserial tab in betaflight configurator.

### Expected behavior
I expected to use all my poorts with the RPM filtering didn't get this errors on my 5 inch quads.

### Flight controller configuration

```
# diff

# version
# Betaflight / STM32F411 (S411) 4.2.5 Nov 22 2020 / 18:38:45 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: BEFH, board_name: BETAFPVF411RX, version: b500c5ae, date: 2020-04-26T06:31:02Z

# start the command batch
batch start

board_name BETAFPVF411RX
manufacturer_id BEFH

# resources
resource SERIAL_TX 2 NONE
resource SERIAL_RX 2 NONE
resource CAMERA_CONTROL 1 A02
resource PINIO 1 A03

# timer
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)

# feature
feature -SOFTSERIAL
feature -RX_SPI
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP

# serial
serial 0 64 115200 57600 0 115200

# led
led 0 0,0::AO:13
led 1 1,0::CB:8
led 2 2,0::CB:8
led 3 3,0::CB:3
led 4 4,0::CB:3

# mode_color
mode_color 6 1 13

# aux
aux 0 0 0 1525 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 35 2 1550 2100 0 0
aux 4 40 3 900 1600 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set gyro_lowpass2_hz = 275
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 220
set dyn_lpf_gyro_max_hz = 550
set acc_calibration = 26,-121,722,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set vbat_max_cell_voltage = 435
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 300
set current_meter = NONE
set ibata_scale = 16000
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2433
set osd_rssi_pos = 2306
set osd_tim_1_pos = 407
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2423
set osd_throttle_pos = 2401
set osd_crosshairs_pos = 2253
set osd_current_pos = 278
set osd_core_temp_pos = 268
set vtx_band = 4
set vtx_channel = 4
set vtx_power = 1
set vtx_freq = 5800
set vcd_video_system = NTSC
set frsky_spi_tx_id = 126,97
set frsky_spi_offset = -31
set frsky_spi_bind_hop_data = 50,56,73,5,11,40,3,22,30,51,37,16,35,62,19,72,65,38,26,36,28,17,4,41,61,15,53,75,20,45,10,31,7,68,49,67,6,48,69,74,58,39,23,9,54,59,70,0,0,0
set frsky_x_rx_num = 1
set camera_control_key_delay = 200
set camera_control_internal_resistance = 99
set camera_control_button_resistance = 125,110,90,75,20
set pinio_box = 40,255,255,255
set gyro_1_sensor_align = CW0FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 0
set gyro_rpm_notch_harmonics = 1

profile 0

# profile 0
set vbat_pid_gain = ON

rateprofile 0

# end the command batch
batch end

# 
```
```
# resource
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource PPM 1 A10
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 11 A08
resource SERIAL_RX 1 A10
resource SERIAL_RX 11 A08
resource LED 1 C14
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource CAMERA_CONTROL 1 A02
resource ADC_BATT 1 A01
resource ADC_CURR 1 B00
resource PINIO 1 A03
resource OSD_CS 1 B12
resource RX_SPI_CS 1 A15
resource RX_SPI_EXTI 1 C13
resource RX_SPI_BIND 1 B02
resource RX_SPI_LED 1 C15
resource RX_SPI_CC2500_TX_EN 1 B09
resource RX_SPI_CC2500_LNA_EN 1 A13
resource RX_SPI_CC2500_ANT_SEL 1 A14
resource GYRO_EXTI 1 B06
resource GYRO_CS 1 A04

Write your command here. Press Tab for AutoComplete.
Save to FileLoad from fileClear output historyCopy to clipboard
Port utilization: D: 0% U: 0%Packet error: 0I2C error: 0Cycle Time: ```

### Setup / Versions
I'm flying a 75mm whoop with the betafpv F4 AIO 1s. I use tx1 and rx1 for the tbs nano reciever. 
Mobula 6 motors and a betafpv camera P01 pro.

### Additional context
dump file included.
[Dump betafpv dump.txt](https://github.com/betaflight/betaflight/files/5740030/Dump.betafpv.dump.txt)

mikeller, 2020-12-24:
Please read and follow the instructions when opening issues. Even after trying 3 times the output of `resource show all` is still missing.

berendb0tje, 2020-12-24:
I'm sorry i tried to do it quick but failed.
I thought i did it wright.

berendb0tje, 2020-12-24:
What did i do wrong?

closed 2020-12-25 by no-response[bot]

mikeller, 2020-12-25:
@berendb0tje:

> What did i do wrong?

=> PASTE resource show all HERE

carloscampos775, 2021-02-07:
I'm using the same board. I have the same issue with Bi-Directional Dshot enabled

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_CURR
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: FREE
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: SERIAL_TX 11
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

mikeller, 2021-02-08:
@carloscampos775: The `resource show all` that you posted has motor output disabled - no surprise that bidirectional Dshot is not working when motor output is disabled. ;-)
If you want to get help for your problem you need to post the required information for a configuration that is _actually showing your problem_.

carloscampos775, 2021-02-10:
@mikeller 
### Describe the bug

No motor output with Bidirectional DShot enabled.

### To Reproduce

Enable Bidirectional DShot no motor output. Disable Bidirectional DShot motor output.

### Expected behavior

Motor output with Bidirectional DShot enabled 

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
# version
# Betaflight / STM32F411 (S411) 4.2.6 Jan  5 2021 / 19:07:43 (a4b6db1e7) MSP API: 1.43
# config: manufacturer_id: BEFH, board_name: BETAFPVF411RX, version: b500c5ae, date: 2020-04-26T06:31:02Z

# start the command batch
batch start

board_name BETAFPVF411RX
manufacturer_id BEFH

# name: CARLOSFPV

# feature
feature -RX_SPI
feature RX_SERIAL

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# aux
aux 0 0 2 1800 2100 0 0
aux 1 1 1 1800 2100 0 0
aux 2 13 0 1400 1600 0 0
aux 3 35 0 1800 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 1000
set acc_calibration = -42,27,-4650,1
set fpv_mix_degrees = 17
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_scale = 111
set yaw_motors_reversed = ON
set small_angle = 180
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_batt_critical = OFF
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2435
set osd_link_quality_pos = 2131
set osd_tim_2_pos = 2451
set osd_craft_name_pos = 2115
set osd_warnings_pos = 14633
set osd_avg_cell_voltage_pos = 2403
set osd_stat_battery = ON
set vtx_band = 4
set vtx_channel = 2
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5760
set name = CARLOSFPV

profile 0

# profile 0
set vbat_pid_gain = ON
set anti_gravity_gain = 5000
set iterm_relax_cutoff = 20
set yaw_lowpass_hz = 100
set p_pitch = 64
set d_pitch = 54
set p_roll = 59
set d_roll = 50
set p_yaw = 63
set angle_level_strength = 70
set d_min_roll = 0
set d_min_pitch = 0

rateprofile 0

# rateprofile 0
set yaw_rc_rate = 152
set yaw_expo = 14
set yaw_srate = 57

# end the command batch
batch end
```
PASTE THE OUTPUT OF 'resource show all' HERE
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_CURR
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: FREE
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: SERIAL_TX 11
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
### Setup / Versions
 <!-- Specify your flight controller model (what type is it, where was it bought from, ...) -->
 - Flight controller: BetaFPV F4 1S AIO FC(SPI Rx)
 <!-- Specify other components attached to the flight controller (RX, VTX, brand / model for all of them, firmware version where applicable...) -->
 - Other components: Crossfire
 <!-- Details about how all is wired -->
 - How are the different components wired up:

Crossfire to UART1

mikeller, 2021-02-10:
@carloscampos775:

> TIM1:
    CH1: SERIAL_TX 11
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

No surprise this is not working for you - there is absolutely no motor output configured. It looks like for some reason your flight controller has lost all configuration. Please reflash it with the latest firmware, and reconfigure it, and then post another `resource show all`.

carloscampos775, 2021-02-10:
@mikeller I will flash the latest firmware again. However, I did this process prior to posting my issue yesterday.

mikeller, 2021-02-10:
@carloscampos775: From the output of `resource show all` that you posted this did not work as intended - make sure you apply your configuration before you test.

carloscampos775, 2021-02-11:
@mikeller I've reflashed and applied my configuration same issue no motor output. I've included the output of `resource show all` with and without Bidirectional Dshot enabled.  When you enable Bidirectional Dshot it disables the timers if you disable Bidirectional Dshot the timers will become active.  

With Bidirectional DShot 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_CURR
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: FREE
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: SERIAL_TX 11
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

Without Bidirectional DShot 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_CURR
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: FREE
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: SERIAL_TX 11
TIM2:
    CH1: MOTOR 2
    CH3: MOTOR 3
TIM3: FREE
TIM4:
    CH2: MOTOR 4
    CH3: MOTOR 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

mikeller, 2021-02-14:
@carloscampos775: Thanks for testing, I was able to reproduce. Looks like detection of resources for bitbanged Dshot is going wrong on this target, and it tries to enable bitbanged Dshot even though the only timer available for it (TIM1) is already used by SOFTSERIAL (which is enabled by default on this target).
Try `set dshot_bitbang = off` - this should be fixing it.

carloscampos775, 2021-02-23:
@mikeller That worked. Thanks for all your work on the project