betaflight/betaflight #10428: CRAZYBEEF4FR unified target - CC2500 not initializing in build #2344 and after

phobos-, 2020-12-27:
### Describe the bug
CC2500 is not initializing in CRAZYBEEF4FR target after PR #10333 has been merged
### To Reproduce
Load build #2344 with full chip erase - CC2500 does not respond, in the configurator receiver is greyed out. RX Led is not blinking. Not able to bind.

### Expected behavior
CC2500 is initialized properly, RX Led is blinking. Able to bind and get rx commands.

### Flight controller configuration
```
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.0 Nov 22 2020 / 09:50:43 (76765cd0e) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: 56f796fb, date: 2019-10-26T09:47:21Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FR
manufacturer_id HAMO
mcu_id 0072003a3439510736383737
signature 

# feature
feature -RX_PARALLEL_PWM

# master
set gyro_1_align_yaw = 900

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 1
TIM3: FREE
TIM4:
    CH1: MOTOR 2
    CH2: MOTOR 3
    CH3: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
Target: CRAZYBEEF4FR
Manufacturer ID: HAMO
Version: 4.3.0 #2344
Binary: betaflight_4.3.0_STM32F411-2344-76765cd0e.hex
Date: 22-11-2020 10:43
Unified Target: HAMO-CRAZYBEEF4FR.config
Date: 2019-10-26T09:47:21Z
Release notes:
Enabled SPI Transaction support for F4 and F7.

### Additional context
None

mikeller, 2021-01-16:
@phobos-: Can you please check if setting `cc2500_spi_chip_detect` to off fixes the problem and makes the SPI RX work again?

knoopx, 2021-01-16:
@mikeller setting cc2500_spi_chip_detect to off makes it work on mine

phobos-, 2021-01-16:
@mikeller  same here, after setting `cc2500_spi_chip_detect = OFF` it appears to work fine, there is no packet loss higher than usual as well.

mikeller, 2021-01-16:
@knoopx, @phobos-: Thanks for testing, that's encouraging news. I don't have access to test hardware right now, but next week I will, and I'll see how we can make the chip detection work again.

PR #10534 (2021-02-10): Fixed detection of CC2500 with SPI transactions enabled.
Fixes #10428.

Test targets:

[betaflight_4.3.0_STM32F745_d6c71a922.zip](https://github.com/betaflight/betaflight/files/5941049/betaflight_4.3.0_STM32F745_d6c71a922.zip)
[betaflight_4.3.0_STM32F7X2_d6c71a922.zip](https://github.com/betaflight/betaflight/files/5941050/betaflight_4.3.0_STM32F7X2_d6c71a922.zip)
[betaflight_4.3.0_STM32F411_d6c71a922.zip](https://github.com/betaflight/betaflight/files/5941051/betaflight_4.3.0_STM32F411_d6c71a922.zip)
[betaflight_4.3.0_STM32F405_d6c71a922.zip](https://github.com/betaflight/betaflight/files/594 [... cut]
closed 2021-02-10 by mikeller