betaflight/betaflight #10435: Smartaudio protocol doesnt work on F7X2

Asizon, 2020-12-31:
### Describe the bug
Smartaudio 1.0 and 2.0 doesnt work in my two F722 flight controllers. This issue is related to the latest bidir drivers changes on https://github.com/betaflight/betaflight/pull/10388
Tried with #2352 build and works as expected, also I have tried reverting changes on avobe mentioned PR and it works as expected.
@tonycake can you take a look on this?  

### To Reproduce
-Flash latest development firmware
-Configure Smartaudio
-Check sensor tab or try changing smartaudio values, sensor tab doesnt recognize Smartaudio device.

### Expected behavior
Smartaudio working as normal

### Flight controller configuration
Used hardware:
Gravity F7 + Smartaudio 2.0 device
AirbotF7 + Smartaudio 1.0 device
```
board_name AIRBOTF7
manufacturer_id AIRB

# name: Asizon

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# serial
serial 0 64 115200 57600 0 115200
serial 2 16384 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# aux
aux 0 0 0 1525 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 7 16 25 40
vtxtable powerlabels 25 200 500 800

# master
set gyro_lowpass2_hz = 425
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_max_hz = 400
set dyn_lpf_gyro_min_hz = 340
set dyn_lpf_gyro_max_hz = 850
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_auto_smoothness = 5
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set dshot_idle_value = 400
set dshot_burst = OFF
set dshot_bidir = ON
set dshot_bitbang = OFF
set motor_pwm_protocol = DSHOT600
set osd_vbat_pos = 6594
set osd_rssi_pos = 6202
set osd_warnings_pos = 14569
set debug_mode = SMARTAUDIO
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 4
set vtx_freq = 5880
set gyro_rpm_notch_q = 1000
set name = Asizon

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 98
set dyn_lpf_dterm_max_hz = 238
set dyn_lpf_dterm_curve_expo = 7
set dterm_lowpass2_hz = 210
set vbat_sag_compensation = 100
set iterm_relax_cutoff = 30
set throttle_boost_cutoff = 25
set f_pitch = 133
set f_roll = 126
set f_yaw = 126
set ff_smooth_factor = 0
set ff_boost = 20
set idle_min_rpm = 15
set dyn_idle_i_gain = 100
set dyn_idle_d_gain = 100

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 13
set pitch_rc_rate = 13
set yaw_rc_rate = 13
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

# end the command batch
batch end
```

### Setup / Versions
Latest 4.3 development build
Latest 10.8.0 Configurator build

tonycake, 2020-12-31:
This is not good. I am wondering if the smartaudio driver was depending on the previous (incorrect) behavior of the UART driver. 

To be honest I don't have any smartaudio products here to test with, wondering if a developer that has been involved in smartaudio can comment, or confirm that it is indeed broken.

Asizon, 2020-12-31:
Maybe @fujin can verify this issue?

limonspb, 2020-12-31:
@Asizon i tried CLRACINGF7, i see Smartaudio values in sensors tab after i turn on blackbox smartaudio:

Here are my diff, dump and timer_show:

[diff_all.txt](https://github.com/betaflight/betaflight/files/5758202/diff_all.txt)
[dump.txt](https://github.com/betaflight/betaflight/files/5758203/dump.txt)
[timer_show.txt](https://github.com/betaflight/betaflight/files/5758204/timer_show.txt)
[image]

limonspb, 2021-01-01:
My friend got CLRACINGF7 + TBS vtx SA2.1, just confirmed with him, SmartAudio is working fine on the latest release.
So maybe it's just TBS vtxes with 2.0 version with F7 boards?

mikeller, 2021-01-01:
@all: I think we need to be very specific about what devices / manufacturers we are talking about - as experience has shown, devices implementing the SmartAudio protocol are widely differing in how the electrical interface for this protocol is done. So what may work with one device may not work with another one (possibly even from the same manufacturer).
@tonycake: Maybe we need to introduce a 'nopull' option to make this work. :disappointed:

igorchin, 2021-01-03:
Hello! I have tried the latest version of 4.3.0 on an Omnibus F7 with the tbs unify pro32 hv and its smartaudio 2.1 and it works perfectly for me.

valdodron, 2021-01-03:
Hello all!
I have made a test for it using this hardware:
HGLRCF722 flight controller
TBS RACE HV SA2.0
Here a two screenshots, SA working screenshot is in 29-11-20 development build, non working SA screenshot is on the latest development build.
[image]
[image]

Asizon, 2021-01-03:
Thanks for testing @igorchin and @valdodron!! Definitelly seems an issue on TBS 2.0 and 1.0 devices. @valdodron can you check your repeated screenshot? You forgot to put non working sa screenshot, thanks!

Asizon, 2021-01-04:
Ok so this is @valdodron screenshot with no working SA:

[image]

tonycake, 2021-01-04:
Mike, I do agree with the 'Nopull' option to let the Smart Audio driver work with the previous (incorrect) behaviour of the serial driver.  It would be a real shame to back out this critical fix, which affects control and telemetry, while we do this though.

mikeller, 2021-01-04:
@tonycake: I am only backing this out for 4.2.6, so we can release something in the short term that will not break a lot of currently working setups.

limonspb, 2021-01-04:
Found an old TBS vtx with SA 2.1 (Unify PRO 5G8 HV Race Edition V2.0, Batch 1001)
FC: Clracing F7, SA port: TX5.
BF 4.2 works fine. Latest master: **no** SA communication.

mikeller, 2021-01-10:
@limonspb: I suspect that this probably isn't down to the protocol implementation, but much rather to the VTX hardware - different VTX models that support the same protocol might have different outcomes with respect to this problem. Part of the problem is that Smart _Audio_ combines audio circuitry with serial circuitry, resulting in non-standard behaviour for both.

limonspb, 2021-01-10:
@mikeller  yeah I agree. Sad, I do not understand this low-level stuff in the firmware. @tonycake, you are the only hope, please :)

Diwellington, 2021-01-10:
Pit mode for akk fx2 ultimate mini international doesn't works

limonspb, 2021-01-10:
> Pit mode for akk fx2 ultimate mini international doesn't works

Please provide more info. What BF versions did you try, what builds, how you set it up.

JyeSmith, 2021-01-11:
I dont know the fine details of the problem, but could it be as simple as adding a 0x00 dummy byte to the beginning of any SA packet?

https://www.team-blacksheep.com/tbs_smartaudio_rev09.pdf

[image]

haslinghuis, 2021-01-11:
4800 baudrate !!! Using two stopbits is already a sign of timer problems reducing payload efficiency to about 72% instead of 80% with 1 stopbit.

mikeller, 2021-01-12:
Good find @JyeSmith - to my recollection this creates problems with other hardware configurations. But I think this is the explanation of why this is breaking now - SmartAudio uses a non-standard variant of a serial interface that is non-inverted but with a pull down when idle - we will need to add this to the serial port options so it can be set when opening a port for SmartAudio.

ClickNciak, 2021-01-15:
> > Pit mode for akk fx2 ultimate mini international doesn't works
> 
> Please provide more info. What BF versions did you try, what builds, how you set it up.

Sorry, i have resolved.
The problem wasn't Betaflight, was the vtx 
because now after the changed, everything works correctly!
Thx anyway ;-)

PR #10562 (2021-03-23): Modify serial drivers to add a workaround for SmartAudio
Fixes #10435.
Smart Audio, on some transmitters, was broken by the fix for #10220.
Smart Audio will now default to the old (bad) behavior where pulldowns
are being used instead of pullups for most BF supported processors when
in Bidirectional, PushPull mode.
closed 2021-03-23 by mikeller