betaflight/betaflight #10437: T-MOTOR FD HD CURRENT SENSING

CyberDreDD, 2020-12-31:
### Describe the bug

### To Reproduce

### Expected behavior

### Flight controller configuration

```
# version
# Betaflight / STM32F405 (S405) 4.2.5 Nov 22 2020 / 18:38:23 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: TMTR, board_name: TMOTORF4, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name TMOTORF4
manufacturer_id TMTR
mcu_id 002c003c464b50092031324d
signature 

# name: Shreddy

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# serial
serial 1 1 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 2 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 3 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1700 2100 0 0
aux 3 27 4 1700 2100 0 0
aux 4 46 0 1700 2100 0 0
aux 5 13 2 900 1300 0 0
aux 6 35 2 1700 2100 0 0

# master
set gyro_lowpass2_hz = 300
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 240
set dyn_lpf_gyro_max_hz = 600
set acc_calibration = -1,10,88,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1015
set max_check = 1990
set rssi_channel = 12
set rssi_scale = 244
set serialrx_provider = CRSF
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 440
set vbat_warning_cell_voltage = 340
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_descent_dist = 100
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_allow_arming_without_fix = ON
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 2
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_launch_control = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 75
set osd_cap_alarm = 2300
set osd_alt_alarm = 0
set osd_vbat_pos = 14754
set osd_rssi_pos = 14381
set osd_flymode_pos = 14806
set osd_current_pos = 482
set osd_mah_drawn_pos = 450
set osd_gps_speed_pos = 14722
set osd_gps_sats_pos = 14370
set osd_home_dir_pos = 14422
set osd_home_dist_pos = 14390
set osd_avg_cell_voltage_pos = 418
set osd_pit_ang_pos = 14828
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set gyro_1_sensor_align = DEFAULT
set gyro_1_align_yaw = 2700
set gyro_rpm_notch_harmonics = 1
set name = Shreddy

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set auto_profile_cell_count = 4

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set motor_output_limit = 85
set auto_profile_cell_count = 6

profile 2

# profile 2
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set auto_profile_cell_count = 3

# restore original profile selection
profile 2

rateprofile 0

# rateprofile 0
set thr_mid = 0
set roll_rc_rate = 180
set pitch_rc_rate = 180
set yaw_rc_rate = 175
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

```
```
A00: FREE
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 2
B08: MOTOR 3
B09: MOTOR 4
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: GYRO_EXTI
C05: LED 1
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FLASH_CS
C09: CAMERA_CONTROL
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: INVERTER 1
C15: INVERTER 5
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH4: CAMERA_CONTROL
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 T-MOTOR F4 HD Bought from HobbyRC UK
 - Flight controller: 
 TBS NANO RX (Latest firmware) TBS M8 GPS, Vifly Finder 2, Caddx Vista and T-Motor F45A v2 All bought from HobbyRC UK
 - Other components:
 
 - How are the different components wired up: Vista is wired from the Air unit cable supplied using uart 2, vifly finder wired to buzzer (5v, Ground and B-), Nano RX wired to uart 6tx/rx and 5v and ground. GPS wired to uart 5tx/rx and the ESC is plug and play

All i can say is before i select a motor protocol, the system reads at 0.5amp, ive tried it with another t-motor FC from my other build too :(

etracer65, 2020-12-31:
Add a 100nF capacitor between the current sensor wire and ground. The current sensor is picking up electrical noise from the DSHOT signal on the ESC signal lines which are close to the current sensor. And contact the manufacturer and complain about their circuit layout.

CyberDreDD, 2021-01-01:
Thankyou so much Guys, Your product is truly wonderful, and I'm sort of glad it wasnt a betaflight problem.
I will contact T-MOTOR and tell them about this, and in the future i will be sticking to hobbywing products.
Thankyou all once again!!

closed 2021-01-01 by CyberDreDD