betaflight/betaflight #10447: GPS_Rescue with switch < gps_rescue_min_dth

druckgott, 2021-01-04:
### Is your feature request related to a problem? Please describe.
If I use GPS-Rescue with a switch and I´m lower then gps_rescue_min_dth (min 50m) away from my home point the quad is droping

### Describe the solution you'd like
If I´m lowen distance then gps_rescue_min_dth, and I aktive the GPS Rescrue with a switch it should not be activated. Otherwise it´s like disarming which I can use. It will be better if the waring sayes your are not to far away.

### Describe alternatives you've considered
If you distance is less then gps_rescue_min_dth it´s not possible to activate this mode maybe.
Or I do not understand why you can set this value not less then 50m because on some field´s it´s hard to find the quad.

etracer65, 2021-01-04:
You can already do this.
```
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
```
This will disable the safety checks for manually activating GPS Rescue (with the GPS RESCUE mode switch). But remember, **you are deactivating safety checks** and this could cause damage or injury if you accidentally activated the mode when close to you or others.

closed 2021-01-04 by etracer65

druckgott, 2021-01-04:
But I don't want to have to disable the checks I want what I can't aktivste it if I change the switch. So hat I get a warning in the osd.
With the option you mentioned as I understand I can aktivste it also if the distance is not reached. 

But it should only get active if all conditions also the distance is reached.

etracer65, 2021-01-04:
The min distance is a safety check. If you want to activate GPS rescue closer than the minimum distance then you must disable the safety checks.

druckgott, 2021-01-04:
Ok what's checks will be deaktivated is there a overview?

druckgott, 2021-01-04:
And if I aktivste it closer it will not drop down it will work expect some safety things are not working?

etracer65, 2021-01-04:
Read the documentation.

https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode