betaflight/betaflight #10449: Kakute F7 mini & Tekko32 F3 45A Mini - Strange current sensor reading

Kabuby, 2021-01-05:
### Describe the bug
I get a strange reading from the current sensor on my new FC. The value is never stable, it keeps jumping between 0 and 5,10...40 A.
I didn't configure anything special, I just wanted to calibrate the sensor and realized that the reading was strange.

### To Reproduce
Simply install Betaflight 4.2.5 and configure the basic parameters to connect it to the remote control.

### Expected behavior
The current value is never stable. A blackbox recording shows the same problem, with the stick at a constant value during flight the current jumps from 0 to 42 A! (see picture below)

### Flight controller configuration

```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.5 Nov 22 2020 / 18:39:53 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7MINIV3, version: 99b66df5, date: 2020-05-19T01:52:24Z

# start the command batch
batch start

board_name KAKUTEF7MINIV3
manufacturer_id HBRO

# name: ---

# feature
feature -AIRMODE
feature -DYNAMIC_FILTER
feature GPS

# beeper
beeper -ON_USB

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 900 1700 0 0
aux 2 2 2 1700 2100 0 0
aux 3 27 5 1300 2100 0 0
aux 4 13 6 1300 2100 0 0
aux 5 26 0 1700 2100 0 0
aux 6 28 2 1300 2100 0 0
aux 7 35 4 1300 2100 0 0
aux 8 36 1 1700 2100 0 0

# master
set acc_calibration = 27,-62,-367,1
set baro_hardware = AUTO
set min_check = 1015
set max_check = 1985
set rssi_channel = 8
set rc_smoothing_auto_smoothness = 20
set serialrx_provider = CRSF
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set align_board_yaw = 180
set ibata_scale = 65
set ibatv_scale = -1
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gyro_1_align_yaw = 2700
set name = ---

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_EXTI
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: GYRO_CS 1
B03: CAMERA_CONTROL
B04: MOTOR 3
B05: MOTOR 4
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: BEEPER
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: LED 1
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 Kakute Tekko ESC Combo
 - Flight controller: 
 Kakute F7 mini V3
 - Other components:
Tekko32 F3 4in1 mini ESC 45A
 - How are the different components wired up:
Cable included.

Blackbox: RC command [throttle] and Amperage
[image]

etracer65, 2021-01-05:
You have set the current and voltage sensor scales incorrectly. Contact the manufacturer of the ESC and flight controller to determine the proper scales to use.

closed 2021-01-05 by etracer65

Kabuby, 2021-01-05:
How come my post is closed?
The manufacturer's default values are:
vbat_scale = 112
ibata_scale = 275

With these values the behavior is exactly the same, I see jumps from 0 to 2 A even when the drone is stationary at the table. The problem is just that of the current that is not stable as in other FC. The modified scale does not change the problem....

etracer65, 2021-01-05:
> The manufacturer's default values are:
> vbat_scale = 112
> ibata_scale = 275

Yes, so why are your settings -1 and 65?

If you still have problems then you have either wired you components incorrectly or you have a hardware problem. In either case this is not the appropriate place to get support.

Kabuby, 2021-01-05:
> > The manufacturer's default values are:
> > vbat_scale = 112
> > ibata_scale = 275
> 
> Yes, so why are your settings -1 and 65?

These values come from the auto calibration in the battery tab.

> If you still have problems then you have either wired you components incorrectly or you have a hardware problem. In either case this is not the appropriate place to get support.

I wasn't looking for support to mount a drone, but was reporting a possible problem with Betaflight configuration or my FC, see for example: #9330 (I did not apply this solution because my FC is different).
I think it's correct to first check the software for problems before soldering capacitors.

Bye