betaflight/betaflight #10452: Frsky Osd video system doesnt work

Asizon, 2021-01-06:
### Describe the bug
Frsky Osd is not working in the latest master. On cli status I can see osd: frskyOsd but seems that is never initialized, osd tab shows no osd detected and also when lipo plugged no osd appears in the googgles.
I have made a test to find where is the bug and comes from here https://github.com/betaflight/betaflight/pull/9981

### To Reproduce
-Flash latest 4.3 build
-Configure uart for FrskyOsd
-Pluggin lipo an see if osd appears, or go to osd tab and no detected osd chip.

### Expected behavior
Frsky Osd working as normal and showing it.

### Flight controller configuration
Im using OMNIBUS F4 FIREWORKS fc with Frsky Osd in uart 3.
```
# version
# Betaflight / STM32F405 (S405) 4.3.0 Jan  5 2021 / 18:27:21 (norevision) MSP API: 1.44
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4FW, version: b0bd8e54, date: 2019-10-20T13:05:07Z

# start the command batch
batch start

board_name OMNIBUSF4FW
manufacturer_id AIRB

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# feature
feature SOFTSERIAL
feature TELEMETRY

# serial
serial 1 2048 115200 57600 0 115200
serial 2 65536 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 0 1525 2100 0 0
aux 1 13 1 1550 2100 0 0
aux 2 35 3 1650 2100 0 0
aux 3 36 2 1600 2075 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 2
vtxtable powervalues 0 1
vtxtable powerlabels 25 200

# master
set gyro_lowpass2_hz = 425
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_max_hz = 400
set dyn_lpf_gyro_min_hz = 340
set dyn_lpf_gyro_max_hz = 850
set acc_hardware = NONE
set acc_calibration = -10,-63,-80,1
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_auto_smoothness = 7
set serialrx_provider = SBUS
set dshot_idle_value = 450
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set align_board_yaw = 90
set pid_process_denom = 2
set osd_vbat_pos = 2369
set osd_rssi_pos = 2074
set osd_craft_name_pos = 9
set osd_compass_bar_pos = 266
set osd_altitude_pos = 55
set osd_power_pos = 330
set osd_warnings_pos = 14601
set osd_esc_tmp_pos = 259
set osd_displayport_device = FRSKYOSD
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 2
set vtx_freq = 5880
set gyro_1_align_yaw = 1800
set gyro_2_align_pitch = 1800
set gyro_rpm_notch_q = 700

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dyn_lpf_dterm_curve_expo = 7
set dterm_lowpass2_hz = 225
set iterm_relax_cutoff = 20
set yaw_lowpass_hz = 100
set throttle_boost = 7
set throttle_boost_cutoff = 25
set f_pitch = 114
set f_roll = 108
set f_yaw = 108
set ff_smooth_factor = 20

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 10
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50

```
```
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=22degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4006, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1
GYRO=MPU6000
OSD: FRSKYOSD
System Uptime: 484 seconds, Current Time: 2021-01-06T07:43:12.643+00:00
CPU:3%, cycle time: 126, GYRO rate: 7936, RX rate: 33, System rate: 10
Voltage: 3 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS NOPREARM CLI MSP RPMFILTER
```

### Setup / Versions
OMNIBUS F4 FIREWORKS Betaflight 4.3
Betaflight Configurator 10.8.0
Frsky Osd version 2.0

mikeller, 2021-01-09:
@Asizon: What is the output of `tasks`.

Asizon, 2021-01-09:
@mikeller here is the output:
```
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms
00 - (         SYSTEM)      9       3       1    0.5%    0.5%         0
01 - (         SYSTEM)    996       6       1    1.0%    0.5%        12
02 - (           GYRO)   8000      27      15   22.1%   12.5%      1775
03 - (         FILTER)   4000      32      22   13.3%    9.3%      1381
04 - (            PID)   4000      82      63   33.3%   25.7%      4084
07 - (             RX)     31      70      46    0.7%    0.6%        24
08 - (         SERIAL)    100  598587       4 5986.3%    0.5%      1088
09 - (       DISPATCH)    995       4       1    0.8%    0.5%        10
10 - (BATTERY_VOLTAGE)     49       7       2    0.5%    0.5%         1
11 - (BATTERY_CURRENT)     49       7       1    0.5%    0.5%         0
12 - ( BATTERY_ALERTS)      5       3       1    0.5%    0.5%         0
13 - (         BEEPER)    100       6       1    0.5%    0.5%         1
20 - (      TELEMETRY)    250      50       0    1.7%    0.0%        18
23 - (            OSD)     59       3       2    0.5%    0.5%         1
25 - (            CMS)     18       3       2    0.5%    0.5%         0
26 - (        VTXCTRL)      5      26       9    0.5%    0.5%         1
27 - (        CAMCTRL)      5       3       0    0.5%    0.0%         0
29 - (    ADCINTERNAL)      2       4       1    0.5%    0.5%         0
RX Check Function                   4       2                         0
Total (excluding SERIAL)                        77.9%   53.6%
```

mikeller, 2021-01-09:
Thanks @Asizon. The OSD task is running as it should, so not sure what is wrong. I'll do some testing in a couple of weeks when I've got access to my FrSky OSD test hardware.

Asizon, 2021-02-09:
Seems that Frsky Osd doesnt have more maintenance and is broken from months in Betaflight 4.3, maybe we should remove support from Betaflight.

daimon82, 2021-03-24:
I have the same problem, my board: Matek F722HD with builtin frsky osd chip.
Everything is working fine on BF 4.2.6 but on the BF4.3 the osd chip is not detected.

Frencchy, 2021-05-03:
I had the same issue with Matek F722-PX and after a few frustrating hours found this :
https://www.rcgroups.com/forums/showthread.php?3526085-Using-OSD-on-New-Matek-F722PX

Basically, enter the following in Betaflight CLI :
serial 5 65536 115200 57600 0 115200
save

Asizon, 2021-05-03:
Sorry @Frencchy  but i think  your solution/issue is not related to this issue, this issues happens on 4.3

chimera15, 2024-06-18:
Same, did anyone find a solution, when I plug in a lipo there's a blue light that flashes then goes out.