betaflight/betaflight #10453: FURYF4OSD dont work with magnetometer of BN-880 HMC5883l I2C

Dam63, 2021-01-06:
### Describe the bug
I just flash the last firmware for my mamba F405 mk2 flight controller (FURYF4OSD)
GPS is ok but the magnetometer (on SDA and SCL) is not recognise
Model : BN-880 (HMC5883l I2C)

### To Reproduce
Flash FURYF4OSD 4.2.6 on mamba F405 mk2
Plug BN880 GPS
Config on UART6
Allow magnetometer on configuration tab

### Expected behavior
Magnetometer should be available in the pictogram list at the top screen

### Flight controller configuration
```
# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.6 Jan  5 2021 / 19:07:23 (a4b6db1e7) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 45b6e90d, date: 2020-11-27T19:02:33Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FURYF4OSD
manufacturer_id DIAT
mcu_id 00380037374d500320313048
signature 

# name: Nazgul 5

# feature
feature GPS

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 2 115200 115200 0 115200

# aux
aux 0 0 6 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 5 1700 2100 0 0
aux 3 35 4 1700 2100 0 0
aux 4 36 5 1700 2100 0 0

# adjrange
adjrange 0 0 0 900 2100 29 0 0 0

# master
set mag_hardware = HMC5883
set baro_hardware = NONE
set rssi_channel = 7
set rssi_src_frame_errors = OFF
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set debug_mode = ADC_INTERNAL
set gyro_1_align_yaw = 1800
set name = Nazgul 5

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

```
# 
# Building AutoComplete Cache ... Done!
# 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH3: MOTOR 3
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 1
TIM3:
    CH3: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: MOTOR 3
DMA2 Stream 7: FREE
```

### Setup / Versions
Manba F405 MK2 from bangood
BN-880 GPS

Configurateur: 10.7.0
Microprogramme: BTFL 4.2.6
Cible: DIAT/FURYF4OSD(STM32F405)

GPS on UART8
GPS magnetometer (SDA and SCL) on SDA and SCL

On BF
[image]

Magnetometer is ok in iNav 2.6
[image]

Conf iNav
```
diff all

# version
# INAV/FURYF4OSD 2.6.0 Dec  4 2020 / 11:22:52 (cada68e4)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

# servo mix

# servo

# safehome

# logic

# gvar

# feature
feature GPS

# beeper

# map

# serial
serial 2 2048 115200 115200 0 115200
serial 5 2 115200 115200 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 110
set gyro_lpf_type = PT1
set gyro_stage2_lowpass_hz = 250
set gyro_stage2_lowpass_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 200
set acc_hardware = MPU6000
set mag_hardware = HMC5883
set baro_hardware = NONE
set pitot_hardware = NONE
set min_check = 1050
set blackbox_rate_denom = 2
set motor_pwm_protocol = DSHOT600
set throttle_idle =  12.000
set applied_defaults = 2
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set mc_airmode_type = THROTTLE_THRESHOLD

# profile
profile 1

set mc_p_pitch = 44
set mc_i_pitch = 60
set mc_d_pitch = 25
set mc_i_roll = 50
set mc_p_yaw = 45
set mc_i_yaw = 70
set dterm_lpf_hz = 100
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 170
set dterm_lpf2_type = PT1
set yaw_lpf_hz = 30
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set setpoint_kalman_enabled = ON
set setpoint_kalman_q = 200
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

# 
```

etracer65, 2021-01-07:
Paste the following in the CLI and it should work:
```
set mag_hardware = AUTO
set mag_bustype = I2C
set mag_i2c_device = 1
save
```

Dam63, 2021-01-07:
Unfortunately not better :
[image]

Dam63, 2021-01-07:
My bad, works great

Need one more full reboot
[image]

Thanks a lot

closed 2021-01-07 by Dam63

mikeller, 2021-01-10:
@Dam63: `mag_i2c_device = 1` defines which of the I2C buses the magnetometer is connected to.