betaflight/betaflight #10455: iflight SucceX-D F7 TwinG V2.1 Stack

fedefpv, 2021-01-07:
### Describe the bug

### To Reproduce

### Expected behavior

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

LA FC/esc non mi rilevano la batteria, pur essendo collegata al drone ma il drone funziona.
Non rileva amperaggio/voltaggio.
Di consegueza neanche betaflight non li rileva e neanche nella OSD.
[image]

Asizon, 2021-01-07:
@fedefpv we are not all Italians, please use english as language. Also you need to complete all issue template, including diff.

fedefpv, 2021-01-07:
Sorry

The FC / ESC do not detect my battery, despite being connected to the drone but the drone works.
It does not detect amperage / voltage.
Consequently even betaflight does not detect them and not even in the OSD

Asizon, 2021-01-07:
@fedefpv To get support for user configuration problems go to https://slack.betaflight.com/

closed 2021-01-08 by no-response[bot]