betaflight/betaflight #10460: Flywoo Goku F411 - Baro & Magnetometer Issue

500mania, 2021-01-08:
### Describe the bug
connected an external BMP280 works well
connected ad external compass (GY-271), the compass works but the baro stop working

### To Reproduce

### Expected behavior

### Flight controller configuration

```
# 

# diff
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F411 (S411) 4.1.1 Nov 15 2019 / 12:55:17 (1e5e3d369) MSP API: 1.42
# manufacturer_id: FLWO   board_name: FLYWOOF411   custom defaults: NO

# start the command batch
batch start

board_name FLYWOOF411
manufacturer_id FLWO

# name: FLYWOO

# resources
resource BEEPER 1 C14
resource MOTOR 1 A08
resource MOTOR 2 B03
resource MOTOR 3 B10
resource MOTOR 4 A15
resource PPM 1 A02
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 11 B06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 11 B07
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C13
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource ADC_BATT 1 B01
resource ADC_CURR 1 A01
resource FLASH_CS 1 B02
resource OSD_CS 1 B12
resource GYRO_EXTI 1 B05
resource GYRO_CS 1 A04
resource USB_DETECT 1 C15

# timer
timer A02 AF3
# pin A02: TIM9 CH1 (AF3)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer A15 AF1
# pin A15: TIM2 CH1 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin A08 1
# pin A08: DMA2 Stream 1 Channel 6
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6

# mixer
mmix 0  1.000 -0.809  0.659 -1.000
mmix 1  1.000 -1.000 -0.659  1.000
mmix 2  1.000  0.809  0.659  1.000
mmix 3  1.000  1.000 -0.659 -1.000

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature OSD

# serial
serial 0 64 115200 57600 0 115200
serial 1 2 115200 9600 0 115200
serial 30 8192 115200 57600 0 115200

# led
led 0 6,6::CO:8
led 1 7,6::CO:8
led 2 8,6::CO:8
led 3 9,6::CO:8

# aux
aux 0 0 0 1375 2100 0 0
aux 1 2 2 1600 2100 0 0
aux 2 13 1 1700 2100 0 0
aux 3 35 3 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5999 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 AIRWAVE  F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 50 100 200 MAX

# rxfail
rxfail 6 s 2000

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 325
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = 158,-33,54
set mag_bustype = I2C
set mag_i2c_device = 1
set baro_bustype = I2C
set baro_i2c_device = 1
set rssi_channel = 12
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set blackbox_device = SPIFLASH
set dshot_burst = ON
set motor_pwm_protocol = DSHOT300
set vbat_min_cell_voltage = 370
set vbat_warning_cell_voltage = 380
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 800
set beeper_inversion = ON
set beeper_od = OFF
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 25
set gps_rescue_throttle_min = 1300
set gps_rescue_throttle_hover = 1400
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 2
set osd_vbat_pos = 2305
set osd_rssi_pos = 2177
set osd_rssi_dbm_pos = 161
set osd_tim_1_pos = 2401
set osd_tim_2_pos = 2369
set osd_flymode_pos = 2241
set osd_throttle_pos = 2361
set osd_vtx_channel_pos = 2209
set osd_current_pos = 2278
set osd_mah_drawn_pos = 2273
set osd_craft_name_pos = 2081
set osd_gps_speed_pos = 2264
set osd_gps_lon_pos = 2124
set osd_gps_lat_pos = 2092
set osd_gps_sats_pos = 2167
set osd_home_dir_pos = 2190
set osd_home_dist_pos = 2200
set osd_flight_dist_pos = 2232
set osd_altitude_pos = 2327
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2337
set osd_disarmed_pos = 2411
set osd_flip_arrow_pos = 2113
set osd_core_temp_pos = 2296
set osd_log_status_pos = 2145
set system_hse_mhz = 8
set vtx_low_power_disarm = ON
set max7456_spi_bus = 2
set dashboard_i2c_bus = 1
set pinio_box = 40,41,255,255
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set gyro_2_spibus = 1
set name = FLYWOO

profile 2

rateprofile 0

# end the command batch
batch end

# 

PASTE THE OUTPUT OF 'resource show all' HERE
```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: LED_STRIP
A01: ADC_CURR
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 1
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: MOTOR 4
B00: FREE
B01: ADC_BATT
B02: FLASH_CS
B03: MOTOR 2
B04: FREE
B05: GYRO_EXTI
B06: SERIAL_TX 11
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: BEEPER
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 1
TIM2:
    CH1: MOTOR 4
    CH2: MOTOR 2
    CH3: MOTOR 3
TIM3: FREE
TIM4:
    CH1: SERIAL_TX 11
    CH2: SERIAL_RX 11
TIM5:
    CH1: LED_STRIP
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:
# 
# version
# Betaflight / STM32F411 (S411) 4.1.1 Nov 15 2019 / 12:55:17 (1e5e3d369) MSP API: 1.42
# manufacturer_id: FLWO   board_name: FLYWOOF411   custom defaults: NO

mikeller, 2021-01-08:
@500mania:

> ###WARNING: NO CUSTOM DEFAULTS FOUND###

The firmware on your flight controller is installed incorrectly. Reinstall it by selecting the correct target (**NOT** STM32F411) in configurator, and installing it, with 'Full Chip Erase' selected.

500mania, 2021-01-08:
> @500mania:
> 
> > ###WARNING: NO CUSTOM DEFAULTS FOUND###
> 
> The firmware on your flight controller is installed incorrectly. Reinstall it by selecting the correct target (**NOT** STM32F411) in configurator, and installing it, with 'Full Chip Erase' selected.
> 
> After this, post another `diff` please.

Here the new diff.
I flashed the same version (4.1.1) and reloaded a "diff all" for my configuration

# 

# diff

# version
# Betaflight / STM32F411 (S411) 4.1.1 Nov 15 2019 / 12:55:17 (1e5e3d369) MSP API: 1.42
# manufacturer_id: FLWO   board_name: FLYWOOF411   custom defaults: YES

# start the command batch
batch start

board_name FLYWOOF411
manufacturer_id FLWO

# name: FLYWOO

# resources
resource MOTOR 2 B03
resource MOTOR 3 B10
resource MOTOR 4 A15
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 11 B06
resource SERIAL_RX 1 A10
resource SERIAL_RX 11 B07
resource ADC_BATT 1 B01
resource GYRO_EXTI 1 B05

# timer
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)

# dma
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6

# mixer
mmix 0  1.000 -0.809  0.659 -1.000
mmix 1  1.000 -1.000 -0.659  1.000
mmix 2  1.000  0.809  0.659  1.000
mmix 3  1.000  1.000 -0.659 -1.000

# feature
feature -RX_PARALLEL_PWM
feature GPS

# serial
serial 0 64 115200 57600 0 115200
serial 1 2 115200 9600 0 115200
serial 30 8192 115200 57600 0 115200

# led
led 0 6,6::CO:8
led 1 7,6::CO:8
led 2 8,6::CO:8
led 3 9,6::CO:8

# aux
aux 0 0 0 1375 2100 0 0
aux 1 2 2 1600 2100 0 0
aux 2 13 1 1700 2100 0 0
aux 3 35 3 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5999 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 AIRWAVE  F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 50 100 200 MAX

# rxfail
rxfail 6 s 2000

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 325
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = 158,-33,54
set rssi_channel = 12
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT300
set vbat_min_cell_voltage = 370
set vbat_warning_cell_voltage = 380
set ibata_scale = 800
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 25
set gps_rescue_throttle_min = 1300
set gps_rescue_throttle_hover = 1400
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 2
set osd_vbat_pos = 2305
set osd_rssi_pos = 2177
set osd_rssi_dbm_pos = 161
set osd_tim_1_pos = 2401
set osd_tim_2_pos = 2369
set osd_flymode_pos = 2241
set osd_throttle_pos = 2361
set osd_vtx_channel_pos = 2209
set osd_current_pos = 2278
set osd_mah_drawn_pos = 2273
set osd_craft_name_pos = 2081
set osd_gps_speed_pos = 2264
set osd_gps_lon_pos = 2124
set osd_gps_lat_pos = 2092
set osd_gps_sats_pos = 2167
set osd_home_dir_pos = 2190
set osd_home_dist_pos = 2200
set osd_flight_dist_pos = 2232
set osd_altitude_pos = 2327
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2337
set osd_disarmed_pos = 2411
set osd_flip_arrow_pos = 2113
set osd_core_temp_pos = 2296
set osd_log_status_pos = 2145
set vtx_low_power_disarm = ON
set gyro_1_align_yaw = 1800
set name = FLYWOO

profile 2

rateprofile 0

# end the command batch
batch end

#

mikeller, 2021-01-09:
@500mania: The configuration looks ok. Can you show a diagram of how you connected the sensors to the I2C bus and where you installed the terminators?

500mania, 2021-01-09:
@mikeller: you think that I need 2 pull-up resitor to put beetween SCL/SDA and 3v3? I assumed that the terminating resistors are implemented in the FC.

mikeller, 2021-01-09:
@500mania: See https://www.ti.com/lit/an/slva689/slva689.pdf for a good paper on I2C terminator use and calculation. The internal pull ups on the STM32 MCUs are on the weak side - they mostly work for on-board components, but if external components are connected an extra set of pull ups is often required. 10M seems to be very high, I'd try something in the 5k - 10k range. But then, if it works for your setup and does not result in excessive I2C errors you can as well leave it as it is. :wink:

500mania, 2021-01-10:
@mikeller: thank you so much. I tried with 1k and 4.7k and....I have no I2C error but...with or whitout Rp! I do not know.... 
the only thing that has changed is that I have moved the compass far away from the barometer.

How can I log I2C error while flying?

mikeller, 2021-01-12:
@500mania: 
> How can I log I2C error while flying?

You can connect the flight controller to configurator after the flight (without disconnecting the battery), and `status` will show the number of I2C errors:

```
[...]
I2C Errors: <x>
[...]
```

closed 2021-02-20 by github-actions[bot]

lucaszanella, 2022-01-08:
Hi @500mania could you tell me how did you flash betaflight into a goku board?