betaflight/betaflight #10475: Use barometer for altitude in the telemetry page open tx

maxgame91, 2021-01-13:
### Describe the bug
Altitude is not display in the telemetry page on open tx when using a barometer only (work with a gps only)

### To Reproduce
A flight controller with a barometer, open tx, and a receiver with telemetry

### Expected behavior
Altitude is display when only a barometer is use in the “ALT” sensors

### Flight controller configuration
Matek F722-SE with a crossfire nano rx and qx7 with opentx
```
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.3 Sep 20 2020 / 20:44:12 (2696b7c88) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722SE, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# feature
feature GPS

# beeper
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# serial
serial 2 2 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# aux
aux 0 0 2 1300 1700 0 0
aux 1 13 3 1300 1700 0 0
aux 2 35 1 1300 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 400 900

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 325
set acc_calibration = 105,45,239,1
set serialrx_provider = CRSF
set dshot_bidir = ON
set yaw_motors_reversed = ON
set small_angle = 180
set gps_auto_baud = ON
set osd_link_quality_pos = 2503
set osd_tim_2_pos = 2369
set osd_throttle_pos = 2306
set osd_mah_drawn_pos = 2228
set osd_gps_speed_pos = 2434
set osd_gps_sats_pos = 2242
set osd_altitude_pos = 2273
set osd_avg_cell_voltage_pos = 2423
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 2
set vtx_freq = 5806
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900
set gyro_rpm_notch_harmonics = 2
set gyro_rpm_notch_min = 125

profile 0

# profile 0
set vbat_sag_compensation = 100
set feedforward_transition = 100
set iterm_relax_cutoff = 10
set throttle_boost = 0
set d_pitch = 32
set f_pitch = 48
set d_roll = 30
set f_roll = 45
set f_yaw = 45
set d_min_roll = 0
set d_min_pitch = 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 142
set pitch_rc_rate = 142
set yaw_rc_rate = 136
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 70
set roll_srate = 63
set pitch_srate = 63
set yaw_srate = 63

# end the command batch
batch end

#
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: SDCARD_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: 
Matel F722-SE on betaflight 4.2.3. Use a crossfire nano rx
 - Other components:
 
 - How are the different components wired up:
Internal baro on the fc and wire for a rx to work

Why the altitude in the barometer is not shown in the telemetry page. I need to have the altitude from the barometer to fly legally. I use a logical function to have an altitude limit for my drone

etracer65, 2021-01-13:
When disarmed the altitude comes from the GPS. So if you have not yet established a fix then the altitude will be zero.

When armed the altitude is a fusion of the barometer and GPS.

This all depends on the `position_alt_source` setting. When set to `AUTO` (the default) it behaves as I've mentioned. You can choose to have the altitude be only from the barometer or GPS if you like.

maxgame91, 2021-01-13:
But the drone without GPS, only a baro,I have nothing in the telemetry altitude. But I have value from my baro on my osd.
I `set position_alt_source = baro_only` but still the same, the altitude on my telemetry page is empty

etracer65, 2021-01-13:
Then you should contact TBS and/or make sure your radio and receivers are flashed with the latest firmware. Betaflight sends the value as I've explained. We have no control over what Crossfire does with the data.

etracer65, 2021-01-13:
Ok, I see what the problem is. Crossfire doesn't have a standalone "altitude" telemetry sensor. It's only included with the GPS data. There is no way to send the altitude separately based on how the CRSF telemetry is defined. So unless the GPS is connected and enabled the frame that contains the GPS data (and altitude) will not be sent.

maxgame91, 2021-01-13:
Thanks for the information, I will contact TBS to see if they can do something, or I will add a gps in all of my build

Thanks for your help

closed 2021-01-13 by maxgame91