betaflight/betaflight #10483: Simplified tuning not applying changes

knoopx, 2021-01-16:
### Describe the bug

Simplified tuning does not apply changes when changing values and pressing "APPLY TUNING". You have to change values, exit simplified tuning menu, re-enter and choose "apply" to make it effective.

"APPLY TUNING" just calls `applySimplifiedTuning` (https://github.com/betaflight/betaflight/blob/master/src/main/cms/cms_menu_imu.c#L294) which relies on `currentPidProfile` that still has unmutated values as these are only updated on exiting (https://github.com/betaflight/betaflight/blob/master/src/main/cms/cms_menu_imu.c#L277).

### To Reproduce

* Enter "SIMPLIFIED TUNING" menu
* Set "PID TUNING" to "RPY"
* Change "MASTER MULT"
* Choose "APPLY TUNING"
* Choose "BACK"
* Enter "PID"
* Values still the same

### Setup / Versions

35de1c822

mikeller, 2021-01-16:
@knoopx: Am I right to assume that this is in CMS (on the OSD screen)?

knoopx, 2021-01-16:
Yes sorry, CMS. I've got a fix but not sure it will be of your liking. Basically I removed the apply menu item and I always apply changes on exit: https://github.com/knoopx/tinyflight/commit/c951b01dc45078e9729c58f69a9d8e8a534456f4

I find that behavior more convenient, If you like that I'll send a proper PR.

mikeller, 2021-01-16:
@knoopx: Yeah, I don't know if we get much value out of giving users the capability to exit the menu without applying the simplified settings, so I think your proposal makes sense. Maybe we should change the label from 'BACK' to 'APPLY'?

knoopx, 2021-02-22:
fixed by #10509

closed 2021-02-22 by knoopx