betaflight/betaflight #10488: Betaflight with MWOSD 1.9.2.0 and not able to bring up BF Config screen (left Stick) on OSD

c172jeff, 2021-01-17:
### Describe the bug

I am running an SPRACINGF4EVO board with Betaflight 4.2.6 installed.  I am also running MWOSD 1.9.2.0 (with Canvas Support compiled in) on a MICRO Minim Osd on UART 3.  This was an upgrade; I previously had this working with earlier versions of the software.  When I power up the install, I see the MWOSD screen on my goggles.  MWOSD is communicating with BetaFlight.  When I move the Drone in the air,  the crosshairs on the MWOSD screen change.
   I can call up the MWOSD config screen via Throttle Half Stick, Right rudder, Pitch up.
   I cannot call up the Betaflight OSD config screen via Throttle Half Stick, Left rudder, Pitch up.
   I believe this used to work.
   In Betaflight, via the configurator, I see the words CMS_MENU when I issue the Throttle Half Stick, Left rudder, Pitch up command, but no OSD config screen appears.

### To Reproduce

Try and bring up the BF Consif screen by using Throttle Half Stick, Left rudder, Pitch up

### Expected behavior

The BF CMS Menu should shown on the MWOSD Screen

### Flight controller configuration

```

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.6 Jan  5 2021 / 19:07:23 (a4b6db1e7) MSP API: 1.43
# config: manufacturer_id: SPRO, board_name: SPRACINGF4EVO, version: 895edcdb, date: 2020-02-02T02:51:32Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name SPRACINGF4EVO
manufacturer_id SPRO
mcu_id 003600223036510135353735
signature 

# name: LilScreamer

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature RSSI_ADC

# beacon
beacon RX_LOST

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200

# aux
aux 0 0 0 1400 2100 0 0
aux 1 13 1 900 1475 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 500

# master
set acc_calibration = -85,-27,-11,1
set mag_hardware = NONE
set serialrx_provider = SBUS
set blackbox_device = NONE
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set current_meter = ADC
set vbat_scale = 112
set ibata_scale = 175
set osd_units = IMPERIAL
set osd_rssi_dbm_pos = 14502
set osd_vtx_channel_pos = 14570
set osd_warnings_pos = 14755
set osd_stick_overlay_left_pos = 6339
set osd_stat_max_alt = ON
set osd_stat_min_rssi_dbm = ON
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5745
set name = LilScreamer

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
```
save
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: LED 1
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: VTX_CS
B00: FREE
B01: FREE
B02: INVERTER 2
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: PREINIT 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_RSSI
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: FREE
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 4
C09: MOTOR 3
C10: FREE
C11: FREE
C12: FREE
C13: GYRO_EXTI
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: ADC 3
DMA2 Stream 2: MOTOR 1
DMA2 Stream 3: MOTOR 2
DMA2 Stream 4: MOTOR 4
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: MOTOR 3

```

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

I am also running MWOSD 1.9.2.0 (with Canvas Support compiled in) on a MICRO Minim Osd on UART 3.  This was an upgrade; I previously had this working with earlier versions of the software.  When I power up the install, I see the MWOSD screen on my goggles.  MWOSD is communicating with BetaFlight.  When I move the Drone in the air,  the crosshairs on the MWOSD screen change.
   I can call up the MWOSD config screen via Throttle Half Stick, Right rudder, Pitch up.
   I cannot call up the Betaflight OSD config screen via Throttle Half Stick, Left rudder, Pitch up.
   I believe this used to work.

   In Betaflight, via the configurator, I see the words CMS_MENU when I issue the Throttle Half Stick, Left rudder, Pitch up command, but no OSD config screen appears.

mikeller, 2021-01-17:
@c172jeff: Setting `sdisplayPortSerial = 3` in CLI should fix this.

c172jeff, 2021-01-17:
That clearly had an effect.

I set 

displayport_msp_serial  = 2 (I am on serial 3 and I assume it is 0 indexed)
and
osd_displayport_device set to MSP

Should I run with OSD in BF Configurator set to on (I guess use bf in canvas mode) or off
use mwosd (which is probably not bad), but bring up bf osd for vtx changes?

I note that when I invoke the bf stick menu, it seems that I cannot fully move thru the menu items shown on the osd.  I mean that when I use the right throttle to move, it sometimes moves up , but not down.  I seems that there is a control problem where I cannot really navigate the menu. It does show and sometimes I move the selection.
I do see it and can move to one item down, sometimes enter a menu, just not in a way that always works?

closed 2021-01-20 by c172jeff