betaflight/betaflight #10492: BMP390 barometer support

sedlacekdavid, 2021-01-18:
Hi Guys, at first, let me thank you for your effort in this project.

Is there any plan to implement support for BMP390 pressure sensor?

Its a very precise sensor with like 0.2m altitude resolution. It might be very useful for return to home safety feature.

Regards, David

closed 2021-01-18 by github-actions[bot]

haslinghuis, 2021-01-18:
Thanks for bringing this up. Sure this will be picked up if it qualifies.

sedlacekdavid, 2021-01-19:
After checking the code it seems that there’s already BMP388 driver in master. Which is excellent news!
BMP390 should be an improved and fully compatible sensor, but I havent tested it yet.

sedlacekdavid, 2021-03-19:
Hi, I have tested it and I just can't get it working. Not sure what else I should try. I got BMP390, communicating over i2c:

My FC is T-motor F7 . I2c communication seems to be working, because I have also a magnetometer on the very same bus and it works correctly.

I also tried to mess around with the parameters but no luck. 

This is the module I have:
https://www.adafruit.com/product/4816

get baro
baro_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE, GYROAUTO
Default value: SPI

baro_spi_device = 0
Allowed range: 0 - 5
Default value: 3

baro_i2c_device = 0
Allowed range: 0 - 5

baro_i2c_address = 119
Allowed range: 0 - 119
Default value: 0

baro_hardware = BMP388
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280, LPS, QMP6988, BMP388
Default value: AUTO

baro_tab_size = 21
Allowed range: 1 - 48

baro_noise_lpf = 600
Allowed range: 0 - 1000

baro_cf_vel = 985
Allowed range: 0 - 1000