betaflight/betaflight #10496: FrSky FPort betaflight telemetry not working

TheIsotopes, 2021-01-19:
### Describe the bug

betaflight telemetry with frsky r-xsr on firmware RXSR_ACCST_2.1.1_Fport_LBT is not working.
i only see the telemetry from the receiver.

i have tested all available firmware fport/sbus, only said firmware does not work.
- RXSR-FPORT_LBT_ACCST_191128 / RXSR_LBT_ACCST_191112 working 
- RXSR_ACCST_2.1.0_LBT working
- RXSR_ACCST_2.1.1_Fport_LBT not working
### To Reproduce

using latest bf 4.3 build configured for fport and frsky r-xsr with firmware RXSR_ACCST_2.1.1_Fport_LBT.frk
### Expected behavior

don't understand why it doesn't work with this firmware.
### Flight controller configuration

```
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jan 17 2021 / 03:12:57 (34614231d) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF722MINI, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF722MINI
manufacturer_id MTKS

# name: S7X2-FPort

# feature
feature -TELEMETRY
feature -OSD

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 1 900 1100 0 0
aux 2 26 2 900 1100 0 0
aux 3 31 0 900 1100 1 0
aux 4 31 3 1900 2100 1 0
aux 5 36 1 1900 2100 0 0

# master
set dyn_lpf_gyro_curve_expo = 7
set acc_hardware = NONE
set rc_smoothing_auto_smoothness = 7
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_sample_rate = 1/8
set ibata_scale = 125
set telemetry_disabled_mode = ON
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
set telemetry_disabled_altitude = ON
set telemetry_disabled_vario = ON
set telemetry_disabled_lat_long = ON
set telemetry_disabled_ground_speed = ON
set telemetry_disabled_distance = ON
set name = S7X2-FPort

# profile 0
set dyn_lpf_dterm_curve_expo = 7
set iterm_relax_cutoff = 20
set throttle_boost = 7
set throttle_boost_cutoff = 25

# rateprofile 0
set rates_type = QUICK
set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 130
set roll_srate = 126
set pitch_srate = 126
set yaw_srate = 99

# end the command batch
batch end

```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller: MATEKF722 Mini
 
 - Other components: RX FrSky R-XSR
 
 - How are the different components wired up:

klutvott123, 2021-01-19:
I just tested the latest master on a matekf722 with R-XSR 2.1.0 ACCST Fport FCC and it works. The release notes says "1.Fix the issue that no output under failsafe mode(F.port firmware only).". It's possible that they broke something in the process. Have you tried 2.1.0 Fport firmware?

TheIsotopes, 2021-01-19:
>  Have you tried 2.1.0 Fport firmware?

I also tested it, forgot to include it.

so do you think it's just a frsky firmware problem?

klutvott123, 2021-01-19:
2.1.0 working and 2.1.1 not working sure makes it look like a RX firmware bug. You could try reporting it here https://github.com/FrSkyRC/Firmware-Test/issues. 🙂

etracer65, 2021-01-19:
Sigh...

From your diff:
```
feature -TELEMETRY
```

klutvott123, 2021-01-19:
You get telemetry with Fport even if it's not enabled.

TheIsotopes, 2021-01-19:
I now also think that this is a frsky problem.
I also have an xm+ that i don't even get bound with the current firmware, a firmware back and it is bound again.

then I'll close here.

closed 2021-01-19 by TheIsotopes

SteveCEvans, 2021-04-01:
See https://github.com/betaflight/betaflight/pull/10656.

klutvott123, 2021-04-03:
Reported to Frsky https://github.com/FrSkyRC/Firmware-Test/issues/118