betaflight/betaflight #10498: Low stick/Arming motor spin up.

qwasert-ben, 2021-01-20:
### Describe the bug
If I have low stick set to 1000, quad will arm and does not control motors.  If I have it set to 1001 or above, quad will arm and motors spin up on when I arm the quad, the spin up is quite low, around 1056 or so.

### To Reproduce
set low stick to 1000, RX won't control motors, set low stick to 1001, motors spin up on arm and then I have contro ofl the motors with the RX.

### Expected behavior
I would like to arm the quad without the motors spinning, I have motor stop set to on.

### Flight controller configuration
'
# Betaflight / STM32F7X2 (S7X2) 4.2.6 Jan  5 2021 / 19:08:42 (a4b6db1e7) MSP API: 1.43
# config: manufacturer_id: TMTR, board_name: TMOTORF7, version: e02dd6f2, date: 2020-11-04T11:31:26Z
'
# start the command batch
batch start
'
board_name TMOTORF7
manufacturer_id TMTR

# name: qav-s sub 250

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# serial
serial 0 64 115200 57600 0 115200

# aux
aux 0 0 0 1875 2100 0 0
aux 1 27 1 1775 2100 0 0

# master
set acc_calibration = 57,-18,-47,1
set min_check = 1020
set serialrx_provider = IBUS
set motor_pwm_protocol = DSHOT150
set pid_process_denom = 2
set name = qav-s sub 250

profile 0

rateprofile 0

# end the command batch
batch end
'
'''
[beta.pdf](https://github.com/betaflight/betaflight/files/5839548/beta.pdf)

PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: LED 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: CAMERA_CONTROL
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: FLASH_CS
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: PINIO 1
C15: PREINIT
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH4: MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: MOTOR 3
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 4
DMA1 Stream 3: FREE
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 3
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
'''

### Setup / Versions
 Tmotor F7 mini 20x20
Tmotor cinematic F30A
RX is FlySky fs-i6x,  Bound to FS-i6 TX
 - Other components:
 <!-- Details about how all is wired -->
I have connected to Bl-heli on my mac and reversed two of the motors and they all spin the correct direction now.  I can't figure out why I arm the quad and the motors spin if I have low stick set to 1001.  If it is set to 1000 the quad will arm and I cannot control the motors with throttle/pitch/yaw/roll.

Any help would be greatly appreciated, I just can't figure it out.
[beta.pdf](https://github.com/betaflight/betaflight/files/5839517/beta.pdf)

qwasert-ben, 2021-01-20:
I have tried to submit this request twice...I don't know what the template wants...

qwasert-ben, 2021-01-20:
I need some help and this is not helping.

mikeller, 2021-01-20:
@qwasert-ben: You are using the wrong ticks (''' instead of \`\`\`) to close your `resource show all` section - GitHub can not deal with this.
Reopening manually.

etracer65, 2021-01-20:
This is not the place to get configuration assistance.

See https://github.com/betaflight/betaflight#support-and-developers-channel

The motors are supposed to spin when you arm and only stop when you disarm. Otherwise anytime you lowered the throttle in flight the motors would stop and the quad will tumble out of control. Using motor stop is NOT recommended.

qwasert-ben, 2021-01-20:
Okay, thank you. Please close thread.

closed 2021-01-20 by etracer65