betaflight/betaflight #10499: Betaflight with MWOSD 1.9.2.0; CMS Menu on the OSD, Stick commands not working to sluggish

c172jeff, 2021-01-20:
### Describe the bug

I had an earlier issue with the CMS menu and am now able to bring up the CMS Menu.
I had to set displayport_msp_serial = 2 (I am on serial 3 and I assume it is 0 indexed)
and osd_displayport_device set to MSP

I can now bring up the CMS Menu on the OSD but the issue is not 100% resolved as it sometimes does not invoke and the other times, the stick commands to navigate the menu have no effect.

The bug is this....

Sometimes it take more than one invocation of Throttle half stick, left rudder, pitch up to bring up the CMS Menu.  In the past, it was an immediate response from the above commands and worked 100% of the time.  It now appears that it works at best 50%. I mean to say that 50% of the time from issuing Throttle half stick, left rudder, pitch up the CMS menu come up on the OSD.  

When the menu does show, I use the sticks to try and move the menu and there is usually/mostly  no movement of the selected CMS item.  I then use both of my transmitter sticks to see if either works.  Hard to say which had an effect, but sometimes this selects an item, enters a menu or moves it to exit.  It does not work as before.

My Sticks are set to 1000  1500, 2000 at the appropriate places .  They are properly received by the flight controller to invoke switches and move the drone, yet in the CMS navigation menu, they seem to be muted 80% at best

### To Reproduce

Just try the above

### Expected behavior

The sticks should navigate the menu and not seem disconnected to non existent.

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```

# diff

# version
# Betaflight / STM32F405 (S405) 4.2.6 Jan  5 2021 / 19:07:23 (a4b6db1e7) MSP API: 1.43
# config: manufacturer_id: SPRO, board_name: SPRACINGF4EVO, version: 895edcdb, date: 2020-02-02T02:51:32Z

# start the command batch
batch start

board_name SPRACINGF4EVO
manufacturer_id SPRO

# name: LilScreamer

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature RSSI_ADC

# beacon
beacon RX_LOST

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200

# aux
aux 0 0 0 1400 2100 0 0
aux 1 13 1 900 1475 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 500

# master
set acc_calibration = -85,-27,-11,1
set mag_hardware = NONE
set serialrx_provider = SBUS
set blackbox_device = NONE
set min_throttle = 1000
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 16000
set align_board_yaw = 180
set current_meter = ADC
set vbat_scale = 112
set ibata_scale = 175
set osd_units = IMPERIAL
set osd_rssi_dbm_pos = 166
set osd_ah_pos = 2126
set osd_warnings_pos = 14472
set osd_stick_overlay_left_pos = 50
set osd_stat_max_alt = ON
set osd_stat_min_rssi_dbm = ON
set osd_displayport_device = MSP
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5745
set displayport_msp_serial = 2
set name = LilScreamer

profile 0

rateprofile 0

# end the command batch
batch end
```
PASTE THE OUTPUT OF 'resource show all' HERE

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: LED 1
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: VTX_CS
B00: FREE
B01: FREE
B02: INVERTER 2
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: PREINIT 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_RSSI
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: FREE
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 4
C09: MOTOR 3
C10: FREE
C11: FREE
C12: FREE
C13: GYRO_EXTI
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: ADC 3
DMA2 Stream 2: MOTOR 1
DMA2 Stream 3: MOTOR 2
DMA2 Stream 4: MOTOR 4
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: MOTOR 3

```

### Setup / Versions
 
 - Flight controller: 
 SPRACINGF4EVO board with Betaflight 4.2.6 installed. I am also running MWOSD 1.9.2.0 (with Canvas Support compiled in) on a MICRO Minim Osd on UART 3.

 
 - Other components:
 
 - How are the different components wired up:

closed 2021-01-21 by c172jeff

mikeller, 2021-01-21:
@c172jeff: What was the solution?

c172jeff, 2021-01-21:
The fix was to use old version of MWOSD that I compiled/flashed a long time ago.  When I compiled flashed the latest version of MWOSD, that seemed to cause problems.  I am using the latest version of BF.  That is not the problem.  The latest version of MWOSD will show an its own OSD (looks nice) that seems to respond to drone movements, it does not work well when BF shows its CMS menu.

etracer65, 2021-01-21:
@c172jeff Perhaps you should open an issue with supporting data to report the bug to the MWOSD project.

c172jeff, 2021-01-21:
I will consider it.  I am gong on vacation so after that, perhaps.  I can also try and compile/flash the new version of MWOSD with varying compile options and determine exactly how I wish to use it.  I am using it on an old drone. The late model drone uses an F7 FC with an osd already in it.  I gain nothing when using MWOSD on the new drone.  For the old drone some questions remain...

1) Use MWOSD and lose rssi display and vtx channel change ability
2) use bf display , which seems to flicker, but get rssi and vtx channel change
3) Spend more time figuring out what to minimally "compile in" to the MWOSD to get it to work however i plan to use it. It did report low memory/stability on the arduino compile

I am good right now.  So, I can use it on my vacation

joetsunami, 2021-01-27:
I am getting the same bug but with a SP Racing F3 Board running BF3.1.7 and MWOSD 1.9.2.0
Like said, running an older version of MWOSD "solves" the problem.