betaflight/betaflight #10500: Profiles with full tuning parameters and switch randomization

SupaflyFPV, 2021-01-21:
### Is your feature request related to a problem? Please describe.

It would be beneficial to allow easy profile based switching of the many parameters which impact flight performance. We can tune in the field to settings we like, and the ability to check and switch easily all relevant parameters would make it easier to do so without accessing a full configurator. 

However, humans are prone to placebo and therefore once preferences are found, it would be beneficial to allow placebo controlled testing by randomising the profiles and recoding the flight and profile via blackbox logging.

As well as supporting pilot driven tuning/optimisation (by reducing going around in circles), it could also improve the quality of evidence for developers when discussing flight related issues, reducing accusations of bias and improving communication. A system which allows pilots to test theories, which gives reliable placebo controlled feedback would help bridge this gap. It may help pilots recognise the problem of placebo, and also convince developers there may be some useful feedback worth acting on to improve flight performance.

### Describe the solution you'd like

All tuning parameter, filter parameters, RC and FF parameters, and TPA in the Profile. This combined with a switch based function to change Profile number, or randomize profile combined with recording a log is the goal.

### Describe alternatives you've considered

Passing the radio etc is not a reliable solution because many pilots fly alone and also - proper test protocol should be at least 5 randomized runs to provide better confidence interval. Technically for good data a pilot must correctly guess 9 out of 10 runs to have a confidence interval of above 95%. However 5 out of 5 is probably sufficient. For those interested in running tests, it would turbocharge the ability to provide useful feedback to have the switching software based

etracer65, 2021-01-21:
Duplicate of https://github.com/betaflight/betaflight/issues/5317

SupaflyFPV, 2021-01-21:
ah yes...great minds...I read the other thread it seemed to have some support...is there a chance this could happen?

etracer65, 2021-01-21:
Since the other issue was created nearly 3 years ago it would seem there's not much interest from anybody in developing this. Besides, once you pull back the covers it gets far more complicated than you're supposing.

SupaflyFPV, 2021-01-21:
I was hoping it might perhaps nicely combine with the activity around 'presets' which I assume also should cover all tuning related settings...that way settings could easily be swapped/shared when all on profile...

ctzsnooze, 2021-01-21:
Perhaps if a smallish subset of settings could be specified, whatever is most important for tuning or testing, eg the RC smoothing parameters, perhaps that might be a feasible and  incremental way to gradually shift some of these things into profiles.
What settings would you find most important to get into the profile setup.
As eTracer says, its a fair bit of work to do this.

SupaflyFPV, 2021-01-22:
How about a two pronged approach? Gyro filters, then RC/assorted?

Batch1:
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 0
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 0
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 400
set dyn_lpf_gyro_curve_expo = 8
set thrust_linear = 0
set gyro_rpm_notch_harmonics = 1
set gyro_rpm_notch_q = 1000
set gyro_rpm_notch_min = 200
set dyn_notch_count = 1
set dyn_notch_bandwidth_hz = 100
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 300

Batch2:
set rc_smoothing_type = FILTER
set rc_smoothing_input_hz = 30
set rc_smoothing_derivative_hz = 30
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set rc_smoothing_auto_smoothness = 10
set dshot_idle_value = 500

closed 2021-02-28 by github-actions[bot]