betaflight/betaflight #10502: Craft ID to follow new European ruling on drones

zmiguel, 2021-01-22:
### Is your feature request related to a problem? Please describe.
This is to request a new OSD element or reconfiguring 'display_name' to allow for usage on the new EU ruling that states drones need to carry their registered pilot id.

### Describe the solution you'd like
1. Add a new OSD element of the type string with at least 24 characters
2. Modify the 'display_name' element to allow at least 24 characters

### Describe alternatives you've considered
As a current alternative, I'm using craft name + display_name to display half of the required string in each and position them together on the osd

---

In Portugal we are required to have a 20 character string that represents the pilot id loaded into the drone software.
Not required to be in the software, but we are also provided with an 18 character unique drone id that could also be placed in software or on the osd, in addition to being physically printed on the drone.

I'm hoping the betaflight team can introduce a place for EU pilots to have their ID in software to be able to easily follow the new laws.

haslinghuis, 2021-01-22:
If this is EU regularity please provide link to document

zmiguel, 2021-01-22:
https://www.easa.europa.eu/faq/116454
https://www.easa.europa.eu/domains/civil-drones/naa
https://www.easa.europa.eu/domains/civil-drones-rpas

Those are the main links with information, I cannot find any examples of IDs anywhere though.
I can only provide the formating for Portugal as that's where I am registered.

zmiguel, 2021-01-22:
Here's an example using my information:

[image]

etracer65, 2021-01-22:
Setting aside the length requirements, there's a fundamental problem that the OSD is only capable of displaying uppercase characters. So we would be unable to accommodate your example of the operator ID.

etracer65, 2021-01-22:
For your example of the drone unique id, if you omit the dashes then it's only 15 characters and the current craft and display name fields accept 16 characters.

zmiguel, 2021-01-22:
I don't think the capitalization would be a problem, all they really want to see is the data there. I could write it on a sticker with my own handwriting and it would still be valid (that's what I did for the attached to the drone part). The issue is the id that needs to be loaded on the drone firmware, either OSD or somewhere else that could be easily accessible.

etracer65, 2021-01-22:
Well clearly the capitalization is important. Your operator ID has both upper and lowercase "r" implying that they are treated differently.

zmiguel, 2021-01-22:
I believe the first 3 characters represent the country, PRT for Portugal. then again I can't find any official documents that specify the id format. I've asked around and Belgium uses BELxxx.

haslinghuis, 2021-01-22:
Cannot find much on `e-ID` yet...

zmiguel, 2021-01-23:
> Those documents are simply TL:DR 😄
> https://www.easa.europa.eu/sites/default/files/dfu/Easy%20Access%20Rules%20for%20Unmanned%20Aircraft%20Systems%20%28Regulations%20%28EU%29%202019-947%20and%20%28EU%29%202019-945%29.pdf

pages 145-147 have the information on the operator ID required to be in the drone

haslinghuis, 2021-01-23:
Hmm overlooked, have seen this pages before but thanks.

haslinghuis, 2021-01-23:
This pages would be a starting point. But how about international and US rules about this,

AndersHoglund, 2021-01-23:
Are there really a need or point in having the Operator Id on OSD?. The EU requirements are:

> As a drone operator flying in the ’open’ category, you must:
>     ensure that the drone displays the drone operator registration number (e.g. with a sticker) and 
>     the same number is uploaded into the remote identification

Ref: [https://www.easa.europa.eu/faq/116467](https://www.easa.europa.eu/faq/116467)
Question is if OSD can be considered "remote identification"?

In Sweden we are only required to have a printed sticker, readable without any aid but normal glasses. An additional QR-code is allowed. No requirement (yet) for any kind of remote identification.

AndersHoglund, 2021-01-23:
I have yet not found the exact EU requirements, but the FAA Remote Id requirements can be found here: [https://tools.ietf.org/id/draft-ietf-drip-reqs-00.html](https://www.faa.gov/uas/getting_started/remote_id/)
Some info from the DRIP wg:[https://tools.ietf.org/id/draft-ietf-drip-reqs-00.html]( https://tools.ietf.org/id/draft-ietf-drip-reqs-00.html)
From what I can see so far, they only talk about remote id via bluetooth and wifi. I.e. RID via a video link (OSD) is not mentioned.

AndersHoglund, 2021-01-23:
Furthermore, there are no RID EU-requirements on C0 (toys <250g) and C4 (home built) drones.  Perfect for BF users.
Good summary here: https://www.noflydrones.co.uk/eu-drone-regulations-blog/2020/6/7/c01234-drone-and-open-category-mapping-summary

haslinghuis, 2021-01-24:
The Toy Provision clause is rather vague. Maybe really small whoops could be classified as toys, not my 250gr AUW 4" drone.

https://translate.google.com/translate?sl=nl&tl=en&u=https://www.dronewatch.nl/2021/01/04/moet-ik-mijn-exploitantnummer-inprogrammeren-in-mijn-drone/

kreidler, 2021-02-05:
> pages 145-147 have the information on the operator ID required to be in the drone

pg. 146:
"The UAS operator must upload the UAS registration number and the three (3) ‘secret digits’ into the
remote identification system of the UAS, if available, or into the electronic-identification system, if
required by the geographical zone."

pg 147:
"upload the full string, which consists of the UAS operator registration number and the
three (3) randomly generated alphanumerics, into the electronic identification system, if
available."

I assume most BF users do not have neither an RID nor an 'electronic identification system' so uploading is not necessary.

mikeller, 2021-02-09:
@zmiguel: To me it seems to be a bit far fetched to assume that OSD equates to the 'remote identification' system that is stipulated by the law. Are there any implementation instructions or court verdicts that indicate that an OSD device meets these requirements?
My assumption is that the purpose of this 'remote identification' requirement is so that data on all UAS flying in certain (restricted) areas can be collected automatically, and their owners held accountable for this - an OSD device seems to be not very suitable for this.

caveymon, 2021-03-04:
Would it be possible to at least have some variable to store the ID in, does not have to be displayed imho. So it can be read when hooked up to betaflight configurator. That way most of the regulations could be satisfied in some form.

haslinghuis, 2021-03-10:
You could use `display_name` even have it in `OSD` if required
```
set display_name = "ID"
```

closed 2021-04-18 by github-actions[bot]