betaflight/betaflight #10504: set stats_min_armed_time_s = 30

MBoerlage, 2021-01-22:
Great new feature in BF 4.3 on total flight stats - (#10215)/, many many thanks to the developers to pick this up. We are already having great fun with it in our chapter - total flight stats is one new bragging right value for us. And with OSD its great for analysis of training DVR. Fun fun fun. 

Just a very minor request to perfection - In the latest implementation default of stats_min_armed_time_s is "-1" which means it not counting flights by default. My request is to "set stats_min_armed_time_s = 30" by default, such that after 30 seconds of armed time, the total flight stats is increased with one flight. 

Why 30 seconds? 
Most successful flights are 18 seconds and longer (I am looking at the multigp UTT leaderboard as reference - the fastest 3 consecutive lap is 18 seconds of air time - add some margin (12s) to that for armed time standing on the block) = 30seconds. If its much less than 30 seconds, there is a big chance that it will count false starts or "arm checks" as flights. 

please let me know if I can further clarify, 

thanks for all the fun you guys keep adding to our hobby!

Matt

limonspb, 2021-01-22:
I agree with 30 seconds. Let's see how PR goes:
https://github.com/betaflight/betaflight/pull/10503

MBoerlage, 2021-01-22:
> I agree with 30 seconds. Let's see how PR goes:
> #10503

you guys are faster than lightning...

etracer65, 2021-01-22:
This is a bad idea. Persistent stats are an optional feature and should not be enabled by default. Very few people actually use this feature. Those that do need to understand the potential conflicts with settings being saved unexpectedly if they use in-flight adjustments. So enabling by default is a very bad idea.

MBoerlage, 2021-01-22:
Thanks for the quick response @etracer65  - I can hardly judge how hard this would press on the available resources in Betaflight, so appreciate that insight.  Up to us to show more people how useful and fun this feature is, and growth that 0.1% you mention, hopefully that will help us show the need for fixing the issue (with whatever more workable solution) in the future.

closed 2021-01-22 by MBoerlage