betaflight/betaflight #10510: rc input extrapolation/prediction (may apply to more inputs)

DEvil0000, 2021-01-24:
### Is your feature request related to a problem? Please describe.
I am aware of rc input interpolation and smoothing features. Also about the fact that filters introduce delay, lost rc frames and not having a fixed update rate.
The obvious question popping to my mind was: why is there no extrapolation method used trying to counter the delay?

### Describe the solution you'd like
The most simple approach would be using euler method on rc input. The more tricky part is how much you wan't to factor it in to compensate for input/filter/mechanical delay. In combination with rc input smoothing this may make up for a great feature. This also gets more interesting since input frames are arriving with higher frequencies for the more modern protocols.

Of course a extrapolation comes not only with computational cost but also the cost of being wrong from time to time. While it should smooth out curves it will be wrong for sharper stick input/changes (or transition from big changes to no changes and the other way around). Sharp input changes however is something we could detect and ignore the extrapolation then for that loop run.

### This may apply to more things
The same principle may also work for gyro input (or maybe better pid input) compensating for filter delay + motor delay or rpm input when controlling the filter position.
If this works out there are of course more advanced extrapolation methods we could use.

### Discussion welcome
I was first thinking about implementing it myself but when being realistic I likely will not have the time to do so. I sadly did fly my quad only one time last year due to the lack of time and other priorities.
In any way I would welcome a discussion about this and want to know your thoughts about this topic.

etracer65, 2021-01-24:
Because extrapolation would lead to overshoot anytime the sticks stopped moving.

DEvil0000, 2021-01-25:
yes, obviously - as I already mentioned. That however does not mean it is worse then delayed input.

etracer65, 2021-01-25:
We've considered this in the past and decided the yes, it's much worse so have rejected this approach. It leads to constant overshoot that is not tunable. Leaving the user unable to determine if their overshoot is due to PID, by RC overshoot, or by how much of either.

The delay in the current implementation is one RC frame at most. If you're concerned about that then you should be using a fast protocol. Crossfire at 150hz is 6.67ms - this is not a noticeable delay.

closed 2021-03-04 by github-actions[bot]